FreeOpcUa/opcua-asyncio #165: Browse the complete opc-ua address space

andreafresa, 2020-03-16:
Hello,
I am trying to retrieve as a client the complete address space of the opc-ua server. The idea should be of having a client cache of the whole address space. Is there a recursive browsing function for exploring the address space? If not, how can I do it? Does someone has already done it? 
Thanks for the help

oroulet, 2020-03-16:
Do not do that. there are 1000s of nodes on servers. 
use paths and get_children() method for nodes you really want.

MD-V, 2020-03-16:
@andreafresa Beware that there could be recursive references!

andrewdodd, 2020-06-02:
Yes, it is unwise to do this blindly as there can be many nodes on a server. However....it is possible to do.

I have a snippet which collects all nodes and their children (NB: it does NOT account for recursive references).

```import asyncio
from datetime import datetime
from asyncua import Client, Node, ua

async def collect_all_under(node, collect_maximum=500000):
    all_nodes = []
    unsearched = [node]

    while unsearched and len(all_nodes) < collect_maximum:
        child = unsearched.pop()
        all_nodes.append(child)
        children = await child.get_children()
        unsearched.extend(children)

    return all_nodes

async def main():
    url = 'opc.tcp://localhost:4840/'
    nodeid = "YOUR NODE ID HERE"
    async with Client(url=url) as client:
        node = client.get_node(nodeid)

        start = datetime.now()
        # pass a number here as 2nd arg to change the max count collected
        nodes_found = await collect_all_under(node)
        duration = datetime.now() - start
        count = len(nodes_found)
        rate = count / duration.total_seconds()

        print("we found {} nodes in {} ({} per second)".format(
            count, duration, rate))
        print("First 10 are:\n" + "\n".join(str(x) for x in nodes_found[:10]))

        return nodes_found

if __name__ == '__main__':
    asyncio.run(main())
```

It is probably not wise to use this in a production setting, but for automating exploratory work it will suffice (It is not very performant).

tukotech, 2020-08-08:
Don't ever ever ever do this on production system, you (can) will be called into "principals" office. 

`import asyncio
import asyncua

from asyncua import Client
import datetime

async def walk(node, reclevel = 0):
    children = await node.get_children()
    
    if children:
        for child in children:
            await walk(child, reclevel+1)
    else:
        value = await node.read_value()
        #try:
            #value = await node.read_value()
        #except asyncua.ua.uatypes.UaStatusCodeError:
            #value = ":bad"
        print('{}:{}'.format(node, value))
        
async def main():
    node_str = '<NodeID>'
    async with Client(url='server_url_here', timeout=60*60) as client:
        node = client.get_node(node_str)
        #node = client.nodes.root

        start = datetime.datetime.now()
        print('Start time : {}'.format(start))
        await walk(node)
        end_time = datetime.datetime.now()
        print('End time : {}'.format(end_time))
        print('Delay {}'.format(end_time - start))
       
            
if __name__ == '__main__':
    asyncio.run(main())
`

andrewdodd, 2020-09-07:
@oroulet  / @tukotech I was wondering if either of you could provide a bit more background why walking a whole system's OPC-UA address space is a bad idea in a production system?

I understand that it would obviously add more load to the CPU, and that it might take a long time if there are lots of nodes to walk...but why would that be a problem?

Is doing this likely to cause a PLC to crash / not meet its SLAs? If this is the case isn't this either:

 * a problem with the OPC-UA server implementation (which should not cause other issues in the PLC), or
 * a more fundamental issue with OPC itself (if there is no way to provide a server implementation that will not cause issues to the PLC)?

Also, considering all of these things, is there a safe way to scan a system in production?

MD-V, 2020-09-07:
@andrewdodd Because there could be circular references between the nodes. If your code doesn't handle circular references you will end up in an inifinite loop...

If you try to browse the address space recursively make sure to specify some cancellation condition (e.g. you browse max 20 layers deep).

andrewdodd, 2020-09-07:
@MD-V Ah, that is a good thing to be aware of definitely. I guess making sure you retain an 'already browsed' list of nodes would solve this too (i.e. a typical DFS or BFS graph traversal would probably work in this situation).

Is there anything else other than being caught in an infinite loop?

swamper123, 2020-09-07:
@andrewdodd What I don't understand is, why you want to archive such a thing? What should be the purpose of that? A server "mirror"/"backup" on client side?

If you got our python-asyncio server, there would be other ways possible (like dumping the addresspace in a pickled file, send the file to client device and load it again (I would not recommened that tbh., but is possible)). 

If you got a PLC OPC-UA Server from a PLC, you can still walk the tree...but you have a pretty bad performance. Meassure the time for a request and count all nodes together, that exist on the server. It would take a while. I haven't worked with PLCs which provide an OPC UA Server, but perhaps you may run into a timing problem between PLC-program and OPC UA Server as well (further informations about that topic are welcomed 😃).

Even if this wouldn't be your problem, there woud still be the problem that you can't just "freeze" the server as well for that time. Your PLC can change values in those _seconds_ you will need to read all values. How would you want to handle if Nodes are deleted while you walk down the whole OPC-UA Server?

andreafresa, 2020-09-07:
You can always browse it and see the address space as a graph. Browse an AS using a DFS like algorithm can solve the recursive references

andrewdodd, 2020-09-07:
@swamper123 I am aware that there is no way to get a snapshot at once of all the nodes...i.e. that if you walk the tree you will get the live value for each node in turn. As far as I can tell though the structure of the tree is not likely to change much (at least not in our PLC).

The reason I am asking is I would like to interrogate the 'shape' of the trees in a variety of deployments (i.e. a "see what tags are present" type of question). Obviously it would be better to do this with design documentation, but at this stage scanning the live system is the only option. 

As far as I can tell though, apart from it being slow there should not be any issue with doing this in a production system?

andreafresa, 2020-09-07:
@swamper123 should the address space in opc-ua be a data model?
If yes, I guess it should be the only way to understand what thing/process the client is trying to control, right?

oroulet, 2020-09-07:
there is not issue browsing for all nodes. It is just that it does not make sense in most cases. I think so far all request to do such things were people not udnerstanding the basis opcua. But of course you may have that rare case when browsing for all nodes is the only solution. Although you should try to be smart and only browse the parts of the address space that you are interested in

andrewdodd, 2020-09-07:
@oroulet Ok, this clarifies it for me. Scanning the whole address space is a bad idea for 'control' and 'timeliness', but for browsing what is available it is the only way. But as you mention, being smart about what you browse will also be helpful.

It is good to know that there are no hidden dragons if you happen to browse everything in a production system.

tukotech, 2020-09-19:
@andrewdodd, sorry for the late reply. 

I have worked with a server with more than 50 million node IDs. If the client is very aggressive enough and speed is fast enough, both the client and the server crashed due to out of memory condition!

What I learned? I am now careful and consider potential impact of browsing the address space.

closed 2020-12-07 by swamper123

manprinsen, 2022-11-06:
hi all!
sorry but i do not agree with you guys who are saying that browsing a opcua server should not be done in production. it is like saying, you should not use for loops in production since they can also "hang" or crash your application...

i think the question is more then valid.
nevertheless here is an example:
```
from opcua import Client
URL = "opc.tcp://100.0.0.3:49320"
try:
    c = Client(URL)
    c.connect()
    print("connected")
except Exception as e:
    print("error",e)

def walk(node, reclevel = 0):
    children = node.get_children()
    
    if children:
        for child in children:
            lt = child.get_display_name()
            name = lt.Text
            if name.startswith("_"):
                continue
            
            try:
                path = str(child.get_path())
                path = path[len("[Node(StringNodeId("):-len("))]")]
                if not path.startswith("ns"):
                    continue
                val = child.get_value()
                dis = child.get_display_name().Text
                print(path, dis, val)
            except Exception as e:
                walk(child, reclevel+1)
            
    else:
        print(str(node))

o = c.get_objects_node()
sim_node = o
#or you can specify the path manually
#c.get_node("ns=2;s=Simulation Examples.Functions")

walk(sim_node)

```
example output:
```
ns=2;s=Simulation Examples.Functions.Ramp1 Ramp1 43
ns=2;s=Simulation Examples.Functions.Ramp2 Ramp2 150.75
ns=2;s=Simulation Examples.Functions.Ramp3 Ramp3 0
ns=2;s=Simulation Examples.Functions.Ramp4 Ramp4 -1000
ns=2;s=Simulation Examples.Functions.Ramp5 Ramp5 -1000000
```

oroulet, 2022-11-06:
It is fine to browse  but browsing the complete address space make no sense at all

andrewdodd, 2022-11-07:
@oroulet you have mentioned that browsing the complete address space makes no sense at all a few times.

Could you provide some more guidance as to why this is the case?

From my understanding there are a few aspects:
 * From a "control" perspective is is unwise, because the underlying process and system will have changed by the time you walk the whole address space, so you will be dealing with stale data.
 * From a "risk" perspective, it could cause problems with the client and/or the server, by causing out-of-memory problems or being caught in a recursive loop.
   * *I think this is indicative of a problem with the programming of either the client, the server, or both. OOM errors and recursive loops are bugs. If the OPC spec and/or standard vendor implementations have problems then they should be fixed, not avoided.*
 * From a "discovery" perspective it is perfectly acceptable.

What different perspectives are there? Can you explain the better alternatives so others can learn from your experience?

AndreasHeine, 2022-11-07:
> From a "control" perspective is is unwise, because the underlying process and system will have changed by the time you walk the whole address space, so you will be dealing with stale data.

you need to rebrowse all at each "BaseModelChangeEvent" or just the changes at each "GeneralModelChangeEvent" so stale data/structure is only a problem if the browse takes to much time

> From a "risk" perspective, it could cause problems with the client and/or the server, by causing out-of-memory problems or being caught in a recursive loop. 

there is a lot of data you dont need to browse like namespace=0 stuff which would make no sence to browse, because every server has it.

> I think this is indicative of a problem with the programming of either the client, the server, or both. OOM errors and recursive loops are bugs. If the OPC spec and/or standard vendor implementations have problems then they should be fixed, not avoided.

recursive references are possible in opc ua but they are very bad practice -> in joint working groups forbidden!

> From a "discovery" perspective it is perfectly acceptable.

i would not use the term "discovery"... in OPC UA it means a own spec. part. i recommend "explore" instead ^^
if you crawl the aspace for specific object-type-instances you already know from a companion spec. yes it makes sense!

andreafresa, 2022-11-07:
Hello guy, sorry to jump in the conversation again. I am now working on different problems and have not been using OPC-UA for a long time, but I think this question is still more than valid. Saying make absolutely no sense means actually that the context is not clear: I was trying to crawl the whole address space as I was working on interoperability between different data and interaction model in the IoT, and because of this I needed to know what was in the address space in order to convert it later to a "Unifying" data model, so the crawling the address space make absolute sense.

About the recursive references, I agree they are a terrible idea, but they are allowed in OPC-UA which means that if you want to create a crawler in OPC-UA Space Address, you need to take care of it and this is a terrible design of OPC-UA to me, they should not be allowed.

Also, from my little experience in OPC-UA I see that if you want to create a Client for a thing that uses OPC-UA, you need to have designed the Address Space of that thing, which to me is a big issue of OPC-UA: in 2022 it is no sense in this decoupling. I think that OPC-UA should consider these problems and solve them. Also,  models of OPC-UA in 2000 lines of XML? Seriously?! Give me a break, please. I think they should really reconsider the design of OPC-UA in general, we do not live in closed worlds and we need easy-to-deploy systems nowadays.

manprinsen, 2022-11-07:
i totally agree with @andreafresa! well spoken!