FreeOpcUa/opcua-asyncio #174: Example code not awaiting

mstfldmr, 2020-04-29:
The sample client codes in examples folder (like client_to_kepware.py, client_to_prosys_cert.py, server-with-encryption.py) are not making asynchronous calls. So, I get errors like:

```
client_to_kepware.py:28: RuntimeWarning: coroutine 'Client.connect' was never awaited
  client.connect()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Root is Node(TwoByteNodeId(i=84))
childs of root are:  <coroutine object Node.get_children at 0x7f3db857a540>
client_to_kepware.py:31: RuntimeWarning: coroutine 'Node.get_children' was never awaited
  print("childs of root are: ", root.get_children())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
name of root is <coroutine object Node.get_browse_name at 0x7f3db857a540>
client_to_kepware.py:32: RuntimeWarning: coroutine 'Node.get_browse_name' was never awaited
  print("name of root is", root.get_browse_name())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
childs og objects are:  <coroutine object Node.get_children at 0x7f3db857a540>
client_to_kepware.py:34: RuntimeWarning: coroutine 'Node.get_children' was never awaited
  print("childs og objects are: ", objects.get_children())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
client_to_kepware.py:38: RuntimeWarning: coroutine 'Node.read_value' was never awaited
  print("tag1 is: {0} with value {1} ".format(tag1, tag1.read_value()))
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
tag1 is: Node(StringNodeId(ns=2;s=Channel1.Device1.Tag1)) with value <coroutine object Node.read_value at 0x7f3db857a540> 
client_to_kepware.py:40: RuntimeWarning: coroutine 'Node.read_value' was never awaited
  print("tag2 is: {0} with value {1} ".format(tag2, tag2.read_value()))
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
tag2 is: Node(StringNodeId(ns=2;s=Channel1.Device1.Tag2)) with value <coroutine object Node.read_value at 0x7f3db857a540> 
client_to_kepware.py:55: RuntimeWarning: coroutine 'Client.disconnect' was never awaited
  client.disconnect()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Traceback (most recent call last):
  File "client_to_kepware.py", line 44, in <module>
    handle1 = sub.subscribe_data_change(tag1)
AttributeError: 'coroutine' object has no attribute 'subscribe_data_change'
sys:1: RuntimeWarning: coroutine 'Client.create_subscription' was never awaited
```

oroulet, 2020-04-30:
Yes some examples are not ported to asyncio. Help welcome!

schroeder-, 2023-03-29:
As I see it they should be fixed

closed 2023-03-29 by schroeder-