FreeOpcUa/opcua-asyncio #180: get_namespace_array is not async

swamper123, 2020-05-05:
Hello together,
just found an Issue here:

https://github.com/FreeOpcUa/opcua-asyncio/blob/5132c4edee2ae8348e5d960a2f890341b899d916/asyncua/server/server.py#L419

I guess `get_namespace_array` shall be a `async def` and return the value of the NamespaceArray Node.

oroulet, 2020-05-05:
yes and there should be a simple test jsut to make sure it does not happen again. Can you send a PR?

swamper123, 2020-05-05:
I will drop sth. in the next days.

swamper123, 2020-05-07:
There have been more files missing an async: 

- get_namespace_array (server.py & client.py)
- import_xml (server.py & client.py)
- delete_nodes (server.py & client.py)
- load_type_definitions (server.py & client.py)
- load_enums (server.py & client.py)
- _add_node (xmlimporter.py)
- stop (history.py in Class HistoryManager seems to be a Coroutine)
- history_read (internal_sessions.py)
- call (internal_sessions.py)

They all expected a Coroutine to return, but these are not awaited. Making a test to check them seems to be more tricky than expected, so this will consume more time than expected.

oroulet, 2020-05-07:
Din noe Try to do everything at once. Choose some simple ones. Also we have tests for xml import. Hope can it be not async?

swamper123, 2020-05-07:
In server.py is written:
https://github.com/FreeOpcUa/opcua-asyncio/blob/5132c4edee2ae8348e5d960a2f890341b899d916/asyncua/server/server.py#L526

If you follow the path to `xmlimporter.import_xml`, you'll see it is a Coroutine. So the server Method shall await this Coroutine and has to be a Coroutine on his own, shall it?

cbergmiller, 2020-05-07:
As long as `Server.import_xml` itself is awaited, it will work (because `Server.import_xml` does return a coroutine from `XmlImporter.import_xml`). But for readability it should probably be a async function.

swamper123, 2020-05-07:
I haven't known this behaviour of coroutines that Coro->Func->Coro is even possible, but well something new learned today. But I would recommened it anyway to change it into a Coro, not just for readability, in case the function may be touched in future, so this would work correctly. But the case `xml_importer` should be in another issue I may do later.

PR #181 (2020-05-07): Add async and await to get_namespace_array methods
Add async and await to get_namespace_array methods
Add /venv to gitignore
Will fix #180
closed 2020-05-07 by oroulet