FreeOpcUa/opcua-asyncio #185: Components in nested object do not appear in OPCUA

comispan, 2020-05-09:
In the code, I had created an AirguardBaseObjectType and AirguardSensorType. The AirguardBaseObjectType is used as a nested object under AirguardSensorType. However, when connected to the server, the variables under AirguardBaseObjectType do not appear (circled in red)?

```
objects = server.get_objects_node()
types = server.get_node(ua.ObjectIds.BaseObjectType)

base = await types.add_object_type(idx, "AirguardBaseObjectType")
await (await base.add_variable(idx, "BadScanCounter", 1, ua.VariantType.Int64)).set_modelling_rule(True)
await (await base.add_variable(idx, "DetectorTemp", 0.0, ua.VariantType.Double)).set_modelling_rule(True)

sensor = await types.add_object_type(idx, "AirguardSensorType")
await (await sensor.add_object(idx, "AirguardBaseObject", objecttype=base)).set_modelling_rule(True)
await (await sensor.add_property(idx, "Latitude", 0.0, ua.VariantType.Double)).set_modelling_rule(True)
await (await sensor.add_property(idx, "Longitude", 0.0, ua.VariantType.Double)).set_modelling_rule(True)
await (await sensor.add_property(idx, "Region", "", ua.VariantType.String)).set_modelling_rule(True)

sensor_1 = await objects.add_object(idx, "Sensor001", objecttype=sensor)
```

[image]

oroulet, 2020-05-11:
Are these child node optional or required in your model?if optionaly ou need to use the instantiate method and make sure to pass the correct optionsvalue

comispan, 2020-05-12:
In the code i have set the modeling rule to true for the nested object. For the nested object type, i have also set the variables modeling rule to true also. Hm so i was thinking that the nested object’s variables would be inherited and instantiated also.

May i know how to call the instantiate method? is there an existing example? thank you!

oroulet, 2020-05-12:
OK. Now I understand. You have an ObjectType inside an ObjectType. The current code skips that kind of things. and honestly I have no idea about the correct behaviour. In my mind it does not make sense, but if someone finds documentation we can change behaviour..

comispan, 2020-05-13:
Hm how would you recommend if we needed to have some common variables
across 2 objects?

oroulet, 2020-05-13:
Inside the new type, create normal variables and objects. They will be instantiated

comispan, 2020-05-13:
Alright thank you for ur help!

closed 2020-05-13 by comispan

cseguna, 2022-04-20:
@oroulet  I have the same issue. I tought this should be standard behaviour that also the nested objects are created correctly.

ahmet-kara, 2024-03-19:
I couldn't understand why this issue is closed.

DavidR921, 2024-04-23:
I would also be interested in how nested object types work.
At least in open62541 it seems to be possible: https://www.open62541.org/doc/master/tutorial_server_object.html#object-types-type-hierarchies-and-instantiation