FreeOpcUa/opcua-asyncio #186: Examples: Event Server Error

FIAV1, 2020-05-12:
Hi, I took the event server example as is and i tried to run it, but all i got was:
```bash
INFO:asyncua.server.internal_session:Created internal session Internal
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15957), requested parent node NumericNodeId(i=11715) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15958), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15959), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15960), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15961), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15962), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15963), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=15964), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=16134), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=16135), requested parent node NumericNodeId(i=15957) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(i=16136), requested parent node NumericNodeId(i=15957) does not exists
WARNING:asyncua.server.server:Endpoints other than open requested but private key and certificate are not set.
INFO:asyncua.server.internal_server:starting internal server
INFO:asyncua.server.binary_server_asyncio:Listening on 0.0.0.0:4840
INFO:asyncua.server.binary_server_asyncio:Closing asyncio socket server
INFO:asyncua.server.internal_server:stopping internal server
INFO:asyncua.server.internal_session:close session Internal
INFO:asyncua.server.subscription_service:delete subscriptions: []
Traceback (most recent call last):
  File "opc-ua-server.py", line 57, in <module>
    asyncio.run(main())
  File "/home/federico/.pyenv/versions/3.8.2/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/home/federico/.pyenv/versions/3.8.2/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "opc-ua-server.py", line 50, in main
    await myevgen.trigger()
TypeError: object NoneType can't be used in 'await' expression
```
I used pip to install the package and I'm running python 3.8.

Am I doing something wrong? Is there a fix to this?

Btw thanks for the hard work 💪

oroulet, 2020-05-12:
there has been a change in master yesterday. Either run opcua-asyncio from master or remove the "await" in the failing line in example code

FIAV1, 2020-05-12:
If I build the package from master branch, do I have this error fixed so i can use await?

oroulet, 2020-05-12:
no need to build but you need to clone then install with for example `pip3 install -e opcua-asyncio`

FIAV1, 2020-05-12:
Ok, removing failing code works correctly, i'll give a try using master branch!

closed 2020-05-12 by FIAV1