FreeOpcUa/opcua-asyncio #188: Server Event JSON serializer

FIAV1, 2020-05-12:
Hi again, I need to find a way to get Event values and types, so i can send them to a Kafka topic.
Is there a way to extract and convert them to a JSON structure?
Or is there any other best practice to do such a job?
Thanks in advance 💪

oroulet, 2020-05-12:
not exactly sure what you want to do but all UA objects should have a ua_types attribute that list all class member and their types. otherwise you can guess types using normal python

FIAV1, 2020-05-12:
Ok i try to put it down better.
I have a server object that triggers two events (i used the example present in this repo); the client subscirbe to this events and prints directly the received event.
Well, in the handler function (client) i need to serialize the event and get keys and values of attributes in it, mutate them to JSON and send them to Kafka.
So my question was: is there a builtin method to serialize the Event data in JSON?

oroulet, 2020-05-12:
No. But it should be easy since all members and types are described in Event.ua_type

FIAV1, 2020-05-12:
Ok, I'll take a look around. Thanks a lot for your time 💪💪

closed 2020-05-12 by FIAV1