FreeOpcUa/opcua-asyncio #189: Connecting Siemens WinCC OPCUA server with Basic128Rsa15 with certificates.

rajnishnationfirst, 2020-05-12:
**Hi ,
I am connecting to WinCC OPCUA Server with Basic128Rsa15 with the below code :**

   ```
  logging.basicConfig(level=logging.WARNING)
    #client = Client("opc.tcp://15.206.2.148:4862")
    client = Client("opc.tcp://EC2AMAZ-VJQV23P:4862")
    client.set_security_string("Basic128Rsa15,SignAndEncrypt,AmazonRootCA1.pem,WinCCVMKey.pem")
    client.application_uri = "urn:EC2AMAZ-VJQV23P:Siemens.Automation.WinCC"
   
    try:
        client.secure_channel_timeout = 30000000
        client.session_timeout = 30000000
        client.connect()
        root = client.get_root_node()
        objects = client.get_objects_node()
        print("childs og objects are: ", objects.get_children())

        #embed()
    finally:
        #client.disconnect()
        print('')

```

But i am getting the below errors:
[image]

Also when i am using the client connection with:
client = Client("opc.tcp://15.206.2.148:4862")  iam getting the TimeOutError.

It would be great if anyone can help me out on this .

Thanks 
Rajnish Vishwakarma

swamper123, 2020-05-13:
Well it says the policie is depricated and don't use it anymore. Maybe this will cause your problem.

myahuang, 2020-08-20:
@swamper123
 If the security policy of latest server library only support  Basic256Sha256 and NoSecurity, other choices such as Basic128Rsa15 and Basic256 are not supported although Class SecurityPolicyType have this two types?

AndreasHeine, 2020-08-20:
@myahuang its not a big deal to reimplement them (for you locally if you use a venv)! i am still searching i had done that for someone in the issues...

AndreasHeine, 2020-08-20:
https://github.com/FreeOpcUa/python-opcua/issues/859

in server.py needs to be added

```python

            if ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic128Rsa15,
                                    ua.MessageSecurityMode.SignAndEncrypt)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic128Rsa15,
                                                               ua.MessageSecurityMode.SignAndEncrypt,
                                                               self.certificate,
                                                               self.private_key)
                                     )
            if ua.SecurityPolicyType.Basic128Rsa15_Sign in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic128Rsa15,
                                    ua.MessageSecurityMode.Sign)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic128Rsa15,
                                                               ua.MessageSecurityMode.Sign,
                                                               self.certificate,
                                                               self.private_key)
                                     )
            if ua.SecurityPolicyType.Basic256_SignAndEncrypt in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic256,
                                    ua.MessageSecurityMode.SignAndEncrypt)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic256,
                                                               ua.MessageSecurityMode.SignAndEncrypt,
                                                               self.certificate,
                                                               self.private_key)
                                     )
            if ua.SecurityPolicyType.Basic256_Sign in self._security_policy:
                self._set_endpoints(security_policies.SecurityPolicyBasic256,
                                    ua.MessageSecurityMode.Sign)
                self._policies.append(ua.SecurityPolicyFactory(security_policies.SecurityPolicyBasic256,
                                                               ua.MessageSecurityMode.Sign,
                                                               self.certificate,
                                                               self.private_key)
                                     )
```

give me a sec. i put it together end post it so you can replace server.py!