FreeOpcUa/opcua-asyncio #191: certificate_handling_failure and *_mismatched_creds TimeourError

swamper123, 2020-05-13:
The two new test in #182 I mentioned here:
https://github.com/FreeOpcUa/opcua-asyncio/pull/182#issuecomment-627897022
seems to fail every time, even if an TimeoutError was thrown.
I am using Python: 3.8.2.

Can anybody reproduce this problem?

swamper123, 2020-05-13:
It seems that `concurrent.futures.TimeoutError` and `asyncio.exceptions.TimeoutError` are not the same.
The tests pass if I replace `TimeoutError` with `asyncio.exceptions.TimeoutError`.
Can anybody reproduce/decline/confirm this?

oroulet, 2020-05-13:
@swamper123 using concurrent.futures in the asyncio code sounds like a bug...maybe do a grep and see ifwe have more..

swamper123, 2020-05-13:
We got some concurrent.future stuff here:
https://github.com/FreeOpcUa/opcua-asyncio/blob/432701845167b2627c46549b4dfdbb46004d8d50/asyncua/server/address_space.py#L468
here:
https://github.com/FreeOpcUa/opcua-asyncio/blob/432701845167b2627c46549b4dfdbb46004d8d50/stdeb.cfg#L4
and in this class:
https://github.com/FreeOpcUa/opcua-asyncio/blob/432701845167b2627c46549b4dfdbb46004d8d50/tests/test_sync.py#L80

I have in my mind, that asyncio and concurrent.futures have/had a common base in something, at least Python 3.6. I don't know if these two worlds are still compatible in newer Python versions.

swamper123, 2020-05-13:
So I found out:

- Python 3.8 has asyncio.exceptions.TimeoutError, which I use.

- Python 3.7 and lower haven't got this module, they uses `concurrent.futures.TimeoutError` directly

This difference between Python 3.8 and Python <=3.7, is annoying, even that `concurrent.futures.TimeoutError` is used in both cases, but in Python 3.8 they raise instead another type of exception:

https://github.com/python/cpython/blob/a63c61168588937c482435d0432c753de4844c46/Lib/asyncio/futures.py#L295

So it seems we have a version-trouble indeed and I haven't got a clue how to deal with that atm.

swamper123, 2020-05-13:
afaik Python 3.7 will get it's final release this summer (just security updates).

In version 3.8 [Happy Eyeballs](https://en.wikipedia.org/wiki/Happy_Eyeballs) is also supported and a good step for the long run with this repo. As well there are many asyncio performaces in general which would be in our favour as well.

swamper123, 2020-07-27:
Seems to be fixed with #209

closed 2020-07-27 by swamper123