FreeOpcUa/opcua-asyncio #192: File transfer from Server to Client

FIAV1, 2020-05-13:
Hi, I consider myself a regular customer now 😄
I saw there's a way to transfer an xml file from server to client, but is there a way to transfer any file from server to client?
Is this considered possible in an OPC UA environment?
I could help developing such a feature if still not present.
Thanks in advance! 💪

swamper123, 2020-05-13:
I think there is a "FileType" in the OPC UA spec (part 5), but I think we haven't got a feature for that so far, have we?

FIAV1, 2020-05-13:
**UPDATE**
I found a workaround, i.e. open a file, read it in binary mode and send it in bytes. 🎉
Question for you guys: is this cool, or could it have performance issue with large files? Is there any better practice?
Thanks 🤓

swamper123, 2020-05-13:
> Question for you guys: is this cool, or could it have performance issue with large files?
Well I would say try it with a large file and do normal other OPC UA stuff concurrent. :^)
I guess it may cause trouble in the end, depending how you send it. Because if you send some GB data in a way, that the whole Eventloop has to wait until transfer is ready, it wouldn'''t be cool.

But what would maybe cool is a possibility to make a "standard" file objekt which contains the "standard" functions.

FIAV1, 2020-05-13:
> I guess it may cause trouble in the end, depending how you send it. Because if you send some GB data in a way, that the whole Eventloop has to wait until transfer is ready, it wouldn'''t be cool.

Ok, this is what i wanted to know, thanks :+1:

> But what would maybe cool is a possibility to make a "standard" file objekt which contains the "standard" functions.

You mean locally, or a new functionality to add to the core?

swamper123, 2020-05-13:
> You mean locally, or a new functionality to add to the core?

I saw [this video](https://www.youtube.com/watch?time_continue=1&v=Jgwt_0y5aCw&feature=emb_logo) and thought about doing such a thing as core function. It is a video of a company, but it seems to be an in general OPC UA thing.

FIAV1, 2020-05-13:
Ok, I watched the video. 

> It is a video of a company, but it seems to be an in general OPC UA thing.

I think it's like you say, so the next question is: both client and server can send a file to each other, or is the communication unidirectional server->client?

swamper123, 2020-05-14:
Well a Server serves what a Client requests, so I would guess it can be possible to send it bidirectional. Problem is, I haven't got the specs (and perhaps for another couple of days \*sigh\*) but will make a proper answer when I got them.

FIAV1, 2020-05-14:
Ok then, I'll wait for updates :+1:

swamper123, 2020-06-22:
It was a long time ago, sty @FIAV1 , but finally I got them...but got also noticed, that you can download them for free on the foundation website, when you create a free account. 🤦 Haven't known that.
On the fly I found Annex C in part 5 (Information Model) which is about File Transfer, but I haven't read it all the way yet. Stay tuned. :D

FIAV1, 2020-06-22:
Hi @swamper123, absolutely no problems, just this morning I thought about this project, I think this is a kind of sign :rofl: 
Anyways, I hope to hear you again soon, I'll stay tuned for sure, thanks a lot :muscle:

swamper123, 2020-06-22:
Hmmm ... I am a bit confused about that stuff.
So yes, it seems like File Transfer is possible. You need open/close methods and read/write methods. Depending where the position of the file handler is, you can read/receive the raw bytes.

So back to your question, yes it is possible, just the implementation will be tricky (or a problem of the opc ua client).

I would recommened you to read [this](https://reference.opcfoundation.org/v104/Core/docs/Part5/C.1/) where you will find the necessarry methods, vars etc. which would be needed to be implemented.
Have a look in the repo, some stuff like: https://github.com/FreeOpcUa/opcua-asyncio/blob/3353bc874c4511dab7187bf8d7614309331cf903/asyncua/ua/uaprotocol_auto.py#L27
Already exist and wait for you to use them. ;)

jadamroth, 2020-12-22:
I created an example in issue #371 if it helps anyone

oroulet, 2020-12-28:
This is now supported . See example. https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-file.py

closed 2020-12-28 by oroulet