FreeOpcUa/opcua-asyncio #193: 0.8.2 -> 0.8.3 Update: Error when connecting a OPC UA Server with Basic256

Take44, 2020-05-18:
After updating opcua-asyncio from 0.8.2 to 0.8.3, the following error occurred when running the program.
The program worked normally before.

'SecurityPolicyBasic256' object has no attribute 'host_certifiate'

What's wrong?

oroulet, 2020-05-18:
It had been renamed to peer or similar. Is that your code or a crash?

Take44, 2020-05-18:
That is my code.

Connecting a OPC UA Server with User/Password/Basic256/SignAndEncrypt/ClientKey/ClientCert(der)/ServerCert(der)

These(User/Password/Basic256/SignAndEncrypt/ClientKey/ClientCert(der)/ServerCert(der)) have not changed anything.

When I switch back to 0.8.2, my code runs fine.

oroulet, 2020-05-18:
What is the stack trace?

Take44, 2020-05-18:
DEPRECATED! Do not use SecurityPolicyBasic256 anymore!
Traceback (most recent call last):
  File "AsyncUAClient.py", line 235, in <module>
    main()
  File "AsyncUAClient.py", line 231, in main
    loop.run_until_complete(task(loop))
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "AsyncUAClient.py", line 143, in task
    await client.connect()
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/client/client.py", line 217, in connect
    await self.open_secure_channel()
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/client/client.py", line 267, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 274, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 204, in open_secure_channel
    self._send_request(request, message_type=ua.MessageType.SecureOpen),
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 129, in _send_request
    msg = self._connection.message_to_binary(binreq, message_type=message_type, request_id=self._request_id)
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/common/connection.py", line 238, in message_to_binary
    chunks = MessageChunk.message_to_chunks(self.security_policy, message, self._max_chunk_size, message_type=message_type, channel_id=self.security_token.ChannelId, request_id=request_id, token_id=self.security_token.TokenId)
  File "/home/User/.pyenv/versions/3.8.2/lib/python3.8/site-packages/asyncua/common/connection.py", line 95, in message_to_chunks
    if security_policy.host_certificate:
AttributeError: 'SecurityPolicyBasic256' object has no attribute 'host_certificate'

Is it your answer?

swamper123, 2020-05-20:
Well I would debug through the `create_session` part and have a look if something is weird, especially at the new lines since 0.8.2
Is there a reason, why `SecurityPolicyBasic256` is marked as deprecated, but is not deleted/commented out?

oroulet, 2020-05-20:
We tried to delete it but many users complained...

swamper123, 2020-05-20:
Okay...weird...but I will keep that in my mind for future issues about that.

ianjohnsonspacex, 2020-10-20:
Was this identified? It seems that neither the SecurityPolicyBasic256 nor SecurityPolicyBasic128Rsa15 policies work, both claiming that the class object has no attribute host_certificate. When I add those attributes to the class constructor, setting them both to None, I get errors indicating that my encryption was rejected.

```
python testerasync.py
DEPRECATED! Do not use SecurityPolicyBasic128Rsa15 anymore!
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:An error occurred verifying security.)
Received an error: MessageAbort(error:StatusCode(BadSecurityChecksFailed), reason:An error occurred verifying security.)
Exception raised while parsing message from server
Traceback (most recent call last):
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 166, in _call_callback
    self._callbackmap[request_id].set_result(body)
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 80, in _process_received_data
    self._process_received_message(msg)
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 99, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 168, in _call_callback
    raise ua.UaError(
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([1])
disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "testerasync.py", line 51, in <module>
    main()
  File "testerasync.py", line 47, in main
    loop.run_until_complete(task(loop))
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "testerasync.py", line 25, in task
    async with client:
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/client.py", line 72, in __aenter__
    await self.connect()
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/client.py", line 244, in connect
    await self.open_secure_channel()
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/client.py", line 294, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 281, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/Users/ijohnson/.venv/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 210, in open_secure_channel
    result = await asyncio.wait_for(
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

```

KishanKishore, 2021-03-11:
Is there an update on this? Are the deprecated policies going to work with ```asyncua==0.9.14``` ?

MD-V, 2021-04-21:
@KishanKishore  @Take44 @ianjohnsonspacex can you please try if this error is fixed in the main branch?