FreeOpcUa/opcua-asyncio #194: Dict as method return value?

swamper123, 2020-05-20:
Is it possible to return a dict as result of a method request?
I tried it with a raw dict, which was rasining an error, but a list was ok.
So down here my example code with my settings:

```
# just a simple example of a nonsense method
@uamethod
async def (my_input_vars):
    d = {my_input_vars.name: "knock knock", "just_a_test":"who is there? ", my_input_vars.name:"Doctor"}
    return d

[...]
ret_val = ua.Argument()
ret_val.Name = "variable results"
ret_val.DataType = ua.NodeId(ua.ObjectIds.String)
ret_val.ValueRank = 1
ret_val.ArrayDimensions = [ ]
ret_val.Description = ua.LocalizedText("Values of Variables")

[...]

await objects.add_method(idx, "get_values", get_device_values, [my_input_vars],[ret_val])

```
Do I need to create an extra extension tyoe or is there a possibility to use a standard one as well?

oroulet, 2020-05-20:
dict is not an ua type so I am not surprised you get issues. send them as json strings

swamper123, 2020-05-20:
This would be a good work around, thx.
So lists are an ua type or are they transformed into an array type?

oroulet, 2020-05-20:
List are not but there was something inside the list that was a known type..

closed 2020-05-20 by swamper123