FreeOpcUa/opcua-asyncio #195: Exception raised while parsing message from server

Take44, 2020-05-25:
When running test for a long time, the following exception occurs. why?
------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 78, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/common/connection.py", line 314, in receive_from_header_and_body
    chunk = MessageChunk.from_header_and_body(self.security_policy, header, body)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/common/connection.py", line 55, in from_header_and_body
    crypto.verify(header_to_binary(obj.MessageHeader) + struct_to_binary(obj.SecurityHeader) + decrypted, signature)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/crypto/security_policies.py", line 161, in verify
    self.Verifier.verify(data, sig)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/crypto/security_policies.py", line 286, in verify
    raise uacrypto.InvalidSignature
AttributeError: module 'asyncua.crypto.uacrypto' has no attribute 'InvalidSignature'

swamper123, 2020-05-25:
Can you give us more backround? How long did you test? What security policy are you using? etc.

Take44, 2020-05-25:
I test it for about 6 hours.
I want to keep it running all the time.

security policy infomation: User/Password/Basic256/SignAndEncrypt/ClientKey/ClientCert(der)/ServerCert(der)

swamper123, 2020-05-25:
If you used `class SecurityPolicyBasic256(SecurityPolicy)` then I won't be able to help, because it is deprecated and shall not be used anymore.
A quick workaround idea would be, that in case the verification fails, it tries to reconnect again.

Take44, 2020-05-25:
The security policy of the actual industrial OPC server currently used supports Basic128Rsa15 and Basic256, but does not support Basic256Sha256.
In such cases, your simple workaround (reconnect) is a good idea, but can't asyncua handle it?

swamper123, 2020-05-25:
1. These encryptions are still in our code and in many other, but are not part of OPC UA anymore since v1.04. 
2. How do you handle other connection errors in general in your code? There can be many reasons why you can't establish a connection in general. This would be one of them for some random reason.
3. We don't know how your code and build-up looks like. Also, since these are deprecated by us and the OPC Foundation, there won't be any "support". It has a reason why it is deprecated.

Take44, 2020-05-25:
I understand.
Thank you very much.
I use your workaround (reconnect).

Take44, 2020-05-29:
For reference, our industrial OPC server has been upgraded to support the Basic256Sha256 security policy.
Therefore, I upgraded the OPC server and tested the same with the Basic256Sha256 security policy, but the same exception occurred.

---
Traceback (most recent call last):
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 78, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/common/connection.py", line 314, in receive_from_header_and_body
    chunk = MessageChunk.from_header_and_body(self.security_policy, header, body)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/common/connection.py", line 55, in from_header_and_body
    crypto.verify(header_to_binary(obj.MessageHeader) + struct_to_binary(obj.SecurityHeader) + decrypted, signature)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/crypto/security_policies.py", line 161, in verify
    self.Verifier.verify(data, sig)
  File "/home/jfeengadmin/.pyenv/versions/3.8.1/lib/python3.8/site-packages/asyncua/crypto/security_policies.py", line 371, in verify
    raise uacrypto.InvalidSignature
AttributeError: module 'asyncua.crypto.uacrypto' has no attribute 'InvalidSignature'

qorinator, 2020-09-10:
Hi I am having similar exception when I run my application for longer than 1 hour this occurs: 

i'm using Basic256Sha256 security option. 
```
Exception raised while parsing message from server
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 79, in _process_received_data
    msg = self._connection.receive_from_header_and_body(header, buf)
  File "/usr/local/lib/python3.8/site-packages/asyncua/common/connection.py", line 323, in receive_from_header_and_body
    self._check_sym_header(security_header)
  File "/usr/local/lib/python3.8/site-packages/asyncua/common/connection.py", line 282, in _check_sym_header
    raise ua.UaError(f"Invalid security token id {security_hdr.TokenId}, expected one of: {expected_tokens}")
asyncua.ua.uaerrors._base.UaError: Invalid security token id 3, expected one of: [1, 2, 0]
```

anyone has any idea what is going on?

swamper123, 2020-09-10:
@qorinator If I remember right, this is the same issue as without security. The client timeouts after an hour. A workaround is to reconnect the Client again, if a timeout happened.

qorinator, 2020-09-10:
ah ok. I'm currently doing that. 
Is there an open issue somewhere I can follow if there is any update regarding this topic? 

Thanks!

swamper123, 2020-09-10:
There was an issue #215 where the problem was handled, but in my opinion isn't resolved properbly. Perhaps take a look at @AndreasHeine Client mentioned there as well. An "Auto Reconnect" would be a nice feature ( see Client Session Services), so if you want to create an PR, go ahead. :)

qorinator, 2020-09-10:
I currently have similar setup for my client as well, so connect -> poll server status -> catch any exceptions and try to reconnect to client. 

But I agree to have an auto-reconnect  would be nice, cause I also saw the discussion regarding _renew_channel_loop, I am not very familiar with opcua behavior in general, but is this an expected behavior? 

where can i find the 'Client Session Services' ?

swamper123, 2020-09-11:
[Here you go. OPC Part 7.5.2 Services, Table 4](https://reference.opcfoundation.org/v104/Core/docs/Part7/5.2/). And here how the specs want that a CLient reestablishes connections ([Part 4, 6.7](https://reference.opcfoundation.org/v104/Core/docs/Part4/6.7/))
This is what the specs say, what a Client should be able to do. As well how to deal with Subscriptions if a reconnection is necessarry.

The solution presented in #215 is a workaround at the moment, but would end up in a state where a full implemented OPC UA Server can end up as well (completly loose Session, create another one and resub to all wanted nodes again, like a "fresh connection"). 
Funfact: In the specs is mentioned that there is even a mechanism where the server reconnects to the Client again and the Client has to do checks, if it is the real server which is reastablishing the new connection. The more you know/read the specs ... 😄

qorinator, 2020-09-11:
😄 ignorance is bliss, fits in this situation haha. 

Not sure if i am qualified to make the PR, since I don't know much about it, but maybe I can play around with it and see how far I get. 

but in theory, it should look like this(at least based on the client re-establishing connection diagram) and just scanning through the repo), right ...?

```python
 try:
    await self.send_hello()
    await self.open_secure_channel() # i assume this is the same as create secure channel or is it not?
except Exception:
    self.disconnect_socket()
    raise

try:
    await self.activate_session(username=self._username, password=self._password,certificate=self.user_certificate)
except FailToAssignNewChannelToExistingSession:
    create_new_session_and_transfer_subscription = True

if create_new_session_and_transfer_subscription:
    await self.create_session()
    await self.activate_session(username=self._username, password=self._password,certificate=self.user_certificate)
    try:
        await client.transfer_subscription_to_new_session() # function doesn't exist yet
    except FailToTransferSubscriptionToNewSession:
        self.create_new_subscription() # or just return and let the calling function deal with creating new subscription

try:
    subscription.re_publish_callback() # not sure if the existing publish_callback can be used
except Bad_SubscriptionIDInvalid:
    self.create_new_subscription() # or just return and let the calling function deal with creating new subscription
except Bad_MessageNotAvailable:
    subscription.publish_callback() # not sure if this is the right function to call for start publish processing
```
what do you think?

swamper123, 2020-09-11:
@qorinator Of course all of us here know all ~1500 pages of Specs and every method in this library. You can wake us up at 3.a.m. for a pop quiz. So better start learning now with page 1, you've got one month! 💪 :trollface: 

Jokes on side, "all creatures are welcomed" and every step further to a full implementation as team is good.

>     await self.open_secure_channel() # i assume this is the same as create secure channel or is it not?

Seems to be the same. While opening a secure_channel, you are creating one.

>         self.create_new_subscription() # or just return and let the calling function deal with creating new subscription

Either we have to store all subscription-references anywhere and call such a method or like you said, return and let the calling function deal with it.

>     subscription.re_publish_callback() # not sure if the existing publish_callback can be used

There is an existing republish method in subscription class, but there is an ToDO inside this method. Perhaps we can fix that in combo with this issue?

> except Bad_MessageNotAvailable:
>     subscription.publish_callback() # not sure if this is the right function to call for start publish processing

This is a Client thing/function_calling thing as well again. Specs say:  _After receiving this status, the Client shall start sending Publish requests with the normal Publish handling. This sequence ensures that the lost NotificationMessages queued in the Server are not overwritten by new Publish responses._
So until the old subscription stuff has finished, be normal again.

qorinator, 2020-09-14:
hahaha  derp. 

i will give it a shot and see what happened, I've been wanting to contribute to an open source project, maybe this could be a start :p

swamper123, 2020-09-15:
You're welcome. :)

I haven't found much more to that topic in the specs. Just that OpenSecureChannel is used to reopen a SecureChannel as well and that subbscriptions have a lifetime...and I also haven't made many thoughts about that topic, because I hadn't got a usecase where the Client was connected for more than 5 min (normally I disconnect).