FreeOpcUa/opcua-asyncio #207: Example: Server Event History DB write Error

araisch, 2020-06-04:
Trying to port the Example for Server Event History:
```python
import asyncio
import logging
from asyncua import ua, Server
from asyncua.server.history_sql import HistorySQLite

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger('asyncua')

async def main():
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:4845/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    # populating our address space
    myobj = await server.nodes.objects.add_object(idx, "MyObject")

    # Creating a custom event
    etype = await server.create_custom_event_type(
        idx, 'MyFirstEvent', ua.ObjectIds.BaseEventType,
        [('MyNumericProperty', ua.VariantType.Float),
        ('MyStringProperty', ua.VariantType.String)])

    # get an event generator for the myobj node which generates custom events
    myevgen = await server.get_event_generator(etype, myobj)
    # Configure server to use sqlite as history database (default is a simple in memory dict)

    sqlite_history = HistorySQLite("my_event_history.sqlite")
    await sqlite_history.init()
    server.iserver.history_manager.set_storage(sqlite_history)

    async with server:
        count = 0
        await server.iserver.enable_history_event(myobj, period=None, count=100)

        while True:
            await asyncio.sleep(5)
            myevgen.event.Severity = count
            myevgen.event.MyStringProperty = "Property %d" % count
            myevgen.event.MyNumericProperty = count
            # generate events for subscribed clients and history
            await myevgen.trigger(message="This is MyFirstEvent %d" % count)
            count += 1

if __name__ == "__main__":
    asyncio.run(main())
```
So it generates a table with the fields of the event with the name 2_1 and then errors:
```
ERROR:aiosqlite:returning exception no such table: 0_2253
Traceback (most recent call last):
  File "\.virtualenvs\opcua-asyncio_TEST-UFX3oYkk\lib\site-packages\aiosqlite\core.py", line 171, in run
    result = function()
sqlite3.OperationalError: no such table: 0_2253
ERROR:asyncua.server.history_sql:Historizing SQL Insert Error for events from NumericNodeId(i=2253): no such table: 0_2253
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='Task-27' coro=<HistorySQLite.save_event() done, defined at \.virtualenvs\opcua-asyncio_TEST-UFX3oYkk\lib\site-packages\asyncua\server\history_sql.py:135> exception=KeyError(NumericNodeId(i=2253))>
Traceback (most recent call last):
  File "\.virtualenvs\opcua-asyncio_TEST-UFX3oYkk\lib\site-packages\asyncua\server\history_sql.py", line 150, in save_event
    period = self._datachanges_period[event.emitting_node]
KeyError: NumericNodeId(i=2253)
```
Is this a bug or am I supposed to generate the table manually?

AndreasHeine, 2020-08-28:
"ns=0;i=2253" => Server-Object-Node looks like your Addressspace is not setup completly

if i comment some of you stuff out its able to start.
```python
import asyncio
import logging
from asyncua import ua, Server
from asyncua.server.history_sql import HistorySQLite

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger('asyncua')

async def main():
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://127.0.0.1:4840/freeopcua/server/")

    # # setup our own namespace, not really necessary but should as spec
    # uri = "http://examples.freeopcua.github.io"
    # idx = await server.register_namespace(uri)

    # # populating our address space
    # myobj = await server.nodes.objects.add_object(idx, "MyObject")

    # # Creating a custom event
    # etype = await server.create_custom_event_type(
    #     idx, 'MyFirstEvent', ua.ObjectIds.BaseEventType,
    #     [('MyNumericProperty', ua.VariantType.Float),
    #     ('MyStringProperty', ua.VariantType.String)])

    # # get an event generator for the myobj node which generates custom events
    # myevgen = await server.get_event_generator(etype, myobj)
    # # Configure server to use sqlite as history database (default is a simple in memory dict)

    # sqlite_history = HistorySQLite("my_event_history.sqlite")
    # await sqlite_history.init()
    # server.iserver.history_manager.set_storage(sqlite_history)

    await server.start()
    # async with server:
    #     count = 0
    #     await server.iserver.enable_history_event(myobj, period=None, count=100)

    #     while True:
    #         await asyncio.sleep(5)
    #         myevgen.event.Severity = count
    #         myevgen.event.MyStringProperty = "Property %d" % count
    #         myevgen.event.MyNumericProperty = count
    #         # generate events for subscribed clients and history
    #         await myevgen.trigger(message="This is MyFirstEvent %d" % count)
    #         count += 1
    while 1:
        await asyncio.sleep(0)

if __name__ == "__main__":
    asyncio.run(main())
```

AndreasHeine, 2020-08-28:
this line breaks it
#await server.iserver.enable_history_event(myobj, period=None, count=100)

AndreasHeine, 2020-08-28:
looks like something is broken...

closed 2023-03-29 by schroeder-