FreeOpcUa/opcua-asyncio #208: How catch error while renewing session Task?

xvax, 2020-06-04:
Hello! 
Dear @oroulet could you please help to understand how it possible to catch this exception?
https://github.com/FreeOpcUa/opcua-asyncio/blob/79072d761e3e1e4fcb00ec324bb6c3f4d13f7dbd/asyncua/client/client.py#L388-L390

I know only method like this
`self.client._renew_channel_task.add_done_callback(self.future_callback_error_logger)`
where in callback call `.result` on Future and handle exceptions, but this is access to protected attribute and feels not right.

First of all issue happens like this

> asyncio Task exception was never retrieved
future: <Task finished name='Task-30774' coro=<Client._renew_channel_loop() done, defined at /usr/local/lib/python3.8/site-packages/asyncua/client/client.py:361> exception=TimeoutError() created at /usr/local/lib/python3.8/site-packages/asyncua/client/client.py:358>
source_traceback: Object created at (most recent call last):
  File "run.py", line 651, in <module>
    p.start()
  File "/usr/local/lib/python3.8/multiprocessing/
...
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/client.py", line 218, in connect
    await self.create_session()
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/client.py", line 358, in create_session
    self._renew_channel_task = self.loop.create_task(self._renew_channel_loop())
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/client.py", line 373, in _renew_channel_loop
    await self.open_secure_channel(renew=True)
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/client.py", line 267, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 271, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/usr/local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 200, in open_secure_channel
    result = await asyncio.wait_for(
  File "/usr/local/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

swamper123, 2020-06-05:
I would say that a try/except block is missing here.
If I understand the[ asyncio-docs ](https://docs.python.org/3/library/asyncio-task.html?highlight=asyncio%20wait_for#asyncio.wait_for)right about `asyncio.wait_for`, the Future will be cancelled, but `asyncio.TimeoutError `is raised before.

oroulet, 2020-06-05:
Renew session is done in the background you cannot catch errors there, but anyway you may get background error due many different uncontrollable things. So the only solution is that I your code you read a node value every 10s or so and disconnect/reconnect if any issues
This is not specific to opcua. You have the same issue with any network protocol

closed 2021-07-19 by swamper123