FreeOpcUa/opcua-asyncio #213: Sync version of Node has no get_properties method

HarrySky, 2020-06-17:
Switched from your `python-opcua` since it is in maintenance mode, planning to go async as soon as possible, but for now I can't yet and use `asyncua.sync` module, which gives me a problem:

`AttributeError: 'Node' object has no attribute 'get_properties'`

Looking at the code I found possible solution: add code below to `asyncua.sync.Node` class

```python
@syncmethod
def get_properties(self):
    return self.get_children(refs=ua.ObjectIds.HasProperty, nodeclassmask=ua.NodeClass.Variable)
```

Can open a Pull Request for this. What do you think?

HarrySky, 2020-06-17:
Patched it up in site-packages. Fix seems to work, but I just got another problem with `get_browse_name` that does not exist in `Node` (but it does exist in original `python-opcua` repository [here](https://github.com/FreeOpcUa/python-opcua/blob/f761ccf0882daff8f000885e851db66b3c0ea167/opcua/common/node.py#L74))

```
[...]
File "[...]asyncua/sync.py", line 70, in wrapper
  aio_func = getattr(self.aio_obj, func.__name__)
AttributeError: 'Node' object has no attribute 'get_browse_name'
```

Is there any compatibility list, or list of things not yet ported? Is it better to stay on `python-opcua` for now?

oroulet, 2020-06-17:
We renamed get_ to read_ in asyncua to be closer to spec

oroulet, 2020-06-17:
Yes. Pr welcome

HarrySky, 2020-06-18:
@oroulet Added PR #216 :smile:

PR #216 (2020-06-18): Add get_properties method to sync version of Node
Fixes #213
closed 2020-06-18 by oroulet