FreeOpcUa/opcua-asyncio #214: server export xml namespace error  and "ns =*;s=*" format

Kowiste, 2020-06-18:
I'm trying to create a OPC server using the[ client-example.py](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client-example.py)

The Server works find and if i try to read using:
`"ns=2;i=3"` or any other i works find, but I can't find the way to read like a `"ns=2;s= ****"`

My client works fine wit a Siemens opc server using `DBName.VariableName` as a s

So I created the xml with `await server.export_xml_by_ns("tr.xml")` to try to understand how to call it. In the xml appear the variable devide_id as `"ns=1;i=3"`
This should be ns 2, right? and  how I could read as a s= ? "Mydevide.device_id". I try it and putting in front the uri but return me StatusBadNodeIDUnknown (0x80340000)`

AndreasHeine, 2020-06-19:
first of all your asking two different questions at once!
await server.export_xml_by_ns() => needs two args XML-File and Namespace or List of Namespaces
1. export xml by ns:

```python
    async def export_xml_by_ns(self, path: str, namespaces: list = None):
        """
        Export nodes of one or more namespaces to an XML file.
        Namespaces used by nodes are always exported for consistency.
        :param path: name of the xml file to write
        :param namespaces: list of string uris or int indexes of the namespace to export, if not provide all ns are used except 0
        """
        if namespaces is None:
            namespaces = []
        nodes = await get_nodes_of_namespace(self, namespaces)
        await self.export_xml(nodes, path)
```

2. Node Identifier String:
in order to use `"ns=2;s= ****"` the variable in the server must be added as so (Identyfier type String)
 `var = await myobj.add_variable(ua.NodeId.from_string(f'ns=2;s=DB_DATA.Test.var'), f'Test.var', 0)`

Kowiste, 2020-06-29:
Thanks work perfect

closed 2020-06-29 by Kowiste