FreeOpcUa/opcua-asyncio #221: get_node via NodeId fails until getting parent first

kjschill, 2020-06-22:
I'm new to OPC-UA so it may be a misunderstanding on how it should work or issue with UA server (FactoryTalk Linx) but if I try to get value of a node via NodeID I get a BadNodeIDUnkown exception from the server. 

If I first interact with the parent of the Node (i.e. call `get_parent_descriptions()`) then call the same code, it works. 

```python
from asyncua.sync import Client, Node, ua

url = "opc.tcp://10.12.27.15:4990/RinggoldUA"
client = Client(url, timeout=60*10)

client.connect()
root = client.nodes.root
objects = client.nodes.objects

#Try to get via NodeId
try:
    value = client.get_node(ua.NodeId("::[CART15]Program:MainProgram.debug_MCPCOMS", 3)).get_value()
    print(value)
except Exception as e:
    print(e)

#Try to get children of parent first
try:
    client.get_node(ua.NodeId("Program:MainProgram", 1465)).get_children_descriptions()
    value = client.get_node(ua.NodeId("::[CART15]Program:MainProgram.debug_MCPCOMS", 3)).get_value()
    print(value)
except Exception as e:
    print(e)

client.disconnect()
```

Response I get is:
`
The node id refers to a node that does not exist in the server address space."(BadNodeIdUnknown)
True`

I was expecting to be able to get get value via NodeId directly.

AndreasHeine, 2020-06-22:
`value = await client.get_value(ns=2;s=DB_DATA.Test.var1004)`
should do the trick

kjschill, 2020-06-22:
I tried switching over to async from the sync version and am still seeing same results.
```python
from asyncua import Client, Node, ua
import asyncio

async def main():
    url = f"opc.tcp://10.12.27.15:4990/RinggoldUA"
    client = Client(url, timeout=60*10)
    await client.connect()
    # root = client.nodes.root
    # objects = client.nodes.objects
    debug_comms = client.get_node("ns=3;s=::[CART17]Program:MainProgram.debug_MCPCOMS")
    #Try to get via NodeId
    try:
        value = await debug_comms.get_value()
        print(value)
    except Exception as e:
        print(e)

    await client.disconnect()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    print('Closing async event loop')
    loop.close()
```
If I then just navigate to parent in UA Expert and re-run code it successfully reads. I don't know if OPC-UA Server then caching something to be know the nodeid.
[image]

Here are the Attributes from UA Expert for the variable I'm trying to read:
[image]

I basically have 40 PLCs with same tag structure and all of them have node  "ns=3;s=::[CART##]Program:MainProgram.debug_MCPCOMS" but the parent node namespace changes for each one.

AndreasHeine, 2020-06-22:
`client.get_node("ns=378;s=Program:MainProgram")`

AndreasHeine, 2020-06-22:
always use this which is in ua expert as nodeid to get the node dont know what this should mean: 

> ::[CART##]

 and 

> .debug_MCPCOMS

[image]

AndreasHeine, 2020-06-22:
> I basically have 40 PLCs with same tag structure and all of them have node "ns=3;s=::[CART##]Program:MainProgram.debug_MCPCOMS" but the parent node namespace changes for each one.

this is like a absolute address so for 40 PLC/servers you would need 40 client connections with the same node!?

You can use the Path aswell if its always the same:
`obj = await client.nodes.root.get_child(["0:Objects", "2:MyObject", "2:Parent"]) `

have you seen the examples?
https://github.com/FreeOpcUa/opcua-asyncio/tree/master/examples

kjschill, 2020-06-22:
@AndreasHeine Thanks for the help. I have seen the examples and been through most.  I'm guessing this is a misunderstanding of OPC-UA and not bug with library. I'm looking for efficient way to feed list of around 100 tags that all PLCs have and retrieve value from each.

The 40 PLCS are all show up in UA Expert as individual nodes under the single connection. I was assuming I could browse using UA Expert to get the nodeIDs and then use those directly to get the values in python. 

I can use the nodeIDs only if I first interact with the parent nodes. I'm seeing that I have to `get_children_descriptions()` for each parent node in order for the `namespace_index` to include the children in the index. If I first read all the children then I can call `await client.get_node(f"ns=3;s=::[CART{cartTest}]Program:MainProgram.debug_MCPCOMS").get_value()` and get a value.

If I don't do this first, I get a `BadNodeIdUnknown` exception. For example, `value = await client.get_node(f"ns=3;s=::[CART22]Program:MainProgram.debug_MCPCOMS").get_value()`

```python
from asyncua import Client, Node, ua
import asyncio

async def main():
    url = f"opc.tcp://10.12.27.15:4990/RinggoldUA"
    client = Client(url, timeout=60*10)
    await client.connect()

    ringgold = client.get_node(ua.StringNodeId("UA_RINGGOLD", 3))
    await ringgold.get_children_descriptions()

    for cartTest in range(1,11):
        # Get Namespace Idx for Each Cart and get list of children
        idx = await client.get_namespace_index(f'UA_RINGGOLD.CART{cartTest}')
        await client.get_node(f"ns={idx};s=CART{cartTest}").get_children_descriptions()

        # Get for online tags for each cart and children descriptions
        idx = await client.get_namespace_index(f'UA_RINGGOLD.CART{cartTest}.Online')
        await client.get_node(f"ns={idx};s=Program:MainProgram").get_children_descriptions()

        # Get for children descriptions for MainProgram
        idx = await client.get_namespace_index(f'UA_RINGGOLD.CART{cartTest}.Online.Program:MainProgram')
        await client.get_node(f"ns={idx};s=Program:MainProgram").get_children_descriptions()
        # Now I can get value for the debug_MCPCOMS tag
        value = await client.get_node(f"ns=3;s=::[CART{cartTest}]Program:MainProgram.debug_MCPCOMS").get_value()
        print(f'Cart {cartTest} debug value is {value}')

    #If i try to go direct to via NodeID from UA Expert it gives BadNodeIdUnknown Exception
    try:
        value = await client.get_node(f"ns=3;s=::[CART22]Program:MainProgram.debug_MCPCOMS").get_value()
        print(f'Cart 22 debug value is {value}')
    except Exception as e:
        print(e)

    await client.disconnect()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    print('Closing async event loop')
    loop.close()
```

Output from script
[image]

Is this expected behavior of OPC-UA?

AndreasHeine, 2020-06-22:
maybe the made a custom type... try `await client.load_type_definitions()` after connecting and try without children description

kjschill, 2020-06-23:
@AndreasHeine Thanks again for help. The `await client.load_type_definitions()` didn't help.

I am using subscriptions in another script where I send out value changes on websockets. If I subscribe to tag `f"ns=3;s=::[CART{cartTest}]Program:MainProgram.debug_MCPCOMS"` in that script, it works fine. After, I can come and run the script where I use get_value() and it works without error.

I have some tags that I want to periodically check value or trigger a `set_value()` from a database change so I was planning to not subscribe to those. This is where I found the problem.

My assumption right now is that it is an issue with Rockwell Automations OPC-UA Server implementation and that after I subscribe or browse the parents, the server cache something so subsequent runs of the script in previous comment works fine.

For future reference here is the script where I subscribe to data first, then I am able directly `get_value()` with the NodeID.

```python
from asyncua import Client, ua, Node
import asyncio
import websockets
import json

queue = asyncio.Queue()

class SubHandler(object):

    def datachange_notification(self, node: Node, val, data):
        timestamp_ns = int(data.monitored_item.Value.SourceTimestamp.timestamp() * 1e9)
        queue.put_nowait({'data': {'timestamp': timestamp_ns, 'value': val, 'node': str(node.nodeid.Identifier)}})

    def event_notification(self, event):
        pass

async def main():
    nodes_list = []

    url = "opc.tcp://10.12.27.15:4990/RinggoldUA"
    client = Client(url, timeout=60*10)
    await client.connect()
    # Build TagNames
    for i in range(38):
        full_tag_name = f"::[CART{i + 1}]Program:MainProgram.debug_MCPCOMS"
        node_id = client.get_node(ua.StringNodeId(full_tag_name, 3))
        nodes_list.append(node_id)
    handler = SubHandler()

    uri = "ws://localhost:8765"
    websocket = await websockets.connect(uri)
    sub = await client.create_subscription(500, handler)
    await sub.subscribe_data_change(nodes_list)

    while True:
        data = await queue.get()
        await websocket.send(json.dumps(data))
        queue.task_done()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(main())
    loop.close()
```

I can close this issue unless you have some ideas on how to further debug.

Ilyazyk, 2020-06-23:
@Kris,
check if the server patch is installed:
1088953 - Patch: OPC UA subscriptions don’t work until tag is browsed,
FactoryTalk Gateway 6.11
<https://rockwellautomation.custhelp.com/app/answers/answer_view/a_id/1088953>

as well as others:
1085673 - FactoryTalk Linx Gateway 6.11.00 (CPR9 SR11.1) Patch TOC
<https://rockwellautomation.custhelp.com/app/answers/answer_view/a_id/1085673>

regards,
Ilya Zykov

AndreasHeine, 2020-06-23:
> @kris,
> check if the server patch is installed:
> 1088953 - Patch: OPC UA subscriptions don’t work until tag is browsed,
> FactoryTalk Gateway 6.11 [...]

thats weird....

> I am using subscriptions in another script where I send out value changes on websockets.

interesting i did that too :) [opcua-sub-to-websocket](https://github.com/AndreasHeine/opcua-sub-to-websocket)

kjschill, 2020-06-23:
@Ilyazyk Thanks! I missed those patches and that fixed it! I'm able to directly get values from NodeId now.

@AndreasHeine Looks like we are on similar development paths. :-) I'm using Django Channels as the websocket server to allow clients to subscribe to different tags. Django Channels receives the data from opcua-asyncio and distributes to appropriate groups. 

Closing issue as was a bug with FactoryTalk Linx Gateway that has been patched.

closed 2020-06-23 by kjschill