FreeOpcUa/opcua-asyncio #222: DataType always Shows String by default

Jayant-Pancholi, 2020-06-23:
I am using OPC UA Expert as an Client.
OPC Server adds Object and variable shows DataType as String after Subscription.
But for same object In the Attribute Panel it shows Float DataType.
While adding variable - My1Var = await My1.add_variable(idx, AddNode[1], dict1[node]['Value'])
How can I explicitly mention the specific DataType to the desired Variable.

AndreasHeine, 2020-06-23:
```
value = 255
servicelevel_node.set_value(ua.DataValue(ua.Variant(value, ua.VariantType.Byte)))
```

Jayant-Pancholi, 2020-06-23:
My1 = await server.nodes.objects.add_object(idx,AddNode[0])
            My1Var = await My1.add_variable(idx, AddNode[1], dict1[node]['Value'])
            value = 255
            My1Var.set_value(ua.DataValue(ua.Variant(value, ua.VariantType.Boolean)))

Please see attachments
[image]

AndreasHeine, 2020-06-23:
`ua.VariantType.Boolean => value 255 ?`

this works:
`float_node = await parameter_obj.add_variable(idx, "float", ua.Variant(0.9, ua.VariantType.Float))`
[image]

swamper123, 2020-06-23:
> this works:
> `float_node = await parameter_obj.add_variable(idx, "float", ua.Variant(0.9, ua.VariantType.Float))`

This is the way for floats. If you want other values, just replace `ua.VariantType.<havealookatallthesedatatypesopcuacanhandle>`. 0.9 is the "start Value" for the server.
The string "float" (second argument) is just the Descriptionname for the Node (not that you may be confused).

AndreasHeine, 2020-06-23:
ok what i want to say is if you want a specific type you have to declare it, otherwise the python datatype got converted in a matching type but maybe not the right one...

@swamper123 : <havealookatallthesedatatypesopcuacanhandle> o_0 hard to read :)

Jayant-Pancholi, 2020-06-23:
@swamper123  In what cases, does it shows dataTypes as Null, even after defining that explicitly
[image]

AndreasHeine, 2020-06-23:
```
    async def add_variable(self, nodeid, bname, val, varianttype=None, datatype=None):
        return await create_variable(self, nodeid, bname, val, varianttype, datatype)
```
if you forget some args in add_variable! None == Null

Jayant-Pancholi, 2020-06-23:
How does SoucreTimestamp & ServerTimeStamp differ, How are they configurable  ?

AndreasHeine, 2020-06-23:
timestamps are part of the datavalue object
ua.DataValue <--- A value with an associated timestamp, and quality.

AndreasHeine, 2020-06-23:
example for a byte:
servicelevel_node.set_value(ua.DataValue(ua.Variant(value, ua.VariantType.Byte))) 
thats why i use this way to set values types, timestamps, status and value are set

Jayant-Pancholi, 2020-06-23:
I get Server timestamp as fixed by default for all obejcts and variables.
Can you show how to set server timestamp ?

AndreasHeine, 2020-06-23:
servertimestamp is part of the internal_server.py wich should have the value from datetime.utcnow()

but its a _ method so you should not use it from outside !!! it should self update !!!
```python
    def _set_current_time(self):
        self.loop.create_task(self.current_time_node.write_value(datetime.utcnow()))
        self.loop.call_later(1, self._set_current_time)
```

https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/server/internal_server.py
so you dont have to set it if you setup your nodes right!

Jayant-Pancholi, 2020-06-23:
Its like i always get this ServerTimestamp as in picture 
[image]

AndreasHeine, 2020-06-23:
what do you expect as timestamp and in what timezone you are at?

Jayant-Pancholi, 2020-06-23:
I am in Indian Standard Time , IST (GMT+5:30)
I just want to know how does this work, moreover when I subscribe to any Object here....Its values keeps on Updating but Source TimeStamp remain same as when the server got started.

AndreasHeine, 2020-06-23:
[image]

Jayant-Pancholi, 2020-06-23:
I have also written some code for this, but in my case, only value is updating.
Can you share the piece of code that you have written

AndreasHeine, 2020-06-23:
```
import asyncio, random
from asyncua import ua, Server

async def main():
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://127.0.0.1:4840/test")
    idx = await server.register_namespace("http://andreas-heine.net/UA")
    obj = server.get_objects_node()
    await server.get_node("ns=0;i=2267").set_value(ua.DataValue(ua.Variant(250, ua.VariantType.Byte)))
    
    """
    setup test namespace
    """
    myobj = await obj.add_object(idx, "myOBJ")
    var_list = []
    for i in range(5000):
        var = await myobj.add_variable(ua.NodeId.from_string(f'ns={idx};s=DB_DATA.Test.var{i}'), f'Test.var{i}', 0)
        await var.set_writable(True)
        var_list.append(var)

    async with server:
        while 1:
            for each in var_list:
                await each.set_value(random.randint(1,100))
            await asyncio.sleep(random.randint(1,100)/100)
            
if __name__ == "__main__":
    asyncio.run(main())
```

closed 2023-03-29 by schroeder-