FreeOpcUa/opcua-asyncio #223: [Confirmity needed] Rpi3 freeze while running pytest

swamper123, 2020-06-23:
Hello together,

so while I couldn't understand #220 I wanted to test the stuff on a fresh pi 3+ with buster, a git clone and python 3.7.3 to be sure it is a python-version problem.

Well now while running pytest my Pi freezes and seems to work with 100% CPU (and is getting pretty hot).
It seems to happen either in `test_history_events` or `test_history_limits`. 
Can somebody may confirm similar problems (or even better a solution or a way how I may log something like this?)?

swamper123, 2020-06-23:
I forgot to mention: While running the tests file by file nothing happens, but running all tests, it causes this problem.

AndreasHeine, 2020-06-23:
some ARM stuff!? maybe its possible to test it in a vm with buster!?

AndreasHeine, 2020-06-23:
after long search i found my pi zero /w installed buster...
python: 3.7.3 ?
asyncua: v... ?
do you just run run-tests.sh?

swamper123, 2020-06-23:
Python Version is 3.7.3
Asyncua Version is master branch.
And yes I just tun run-tests.sh (which is nothing else as executing pytest).
No other python versions are existing.

I am setting up a VM to check if this happens there as well.

swamper123, 2020-06-23:
So test result for 3.7.3, raspbian buster on VM _*\*drumm roll\**_ ... he stucks in test_history_eventy.py.

swamper123, 2020-06-23:
😕 ok ... after reboot and just executing the pytest it runs until test_sync and the history tests pass... \*_table flip_\*
I smell thread/background Voodo anywhere, but I am not sure if it is just me. =/

This would explain why a thread is still running while debugging on Ubuntu 20.04 and pycharm...and why some tests fail while testing in a bundle while testing them one by one will let them pass.

AndreasHeine, 2020-06-23:
yeah helped a litte with doing nothing :D

swamper123, 2020-06-23:
If you could test it with your pi zero, this would be realy helpfull. >_<

swamper123, 2020-06-23:
So while starting with CLI, giving Buster one more core and one more GB RAM (2processors and 2 GB RAM, which shouldn't be necessary in my opinion) the tests seem to work fine. I'm not so familiar with VMs and their behaviour compared to a hardware device, but this is strange.

Could it be the lack of ressources that cause this trouble on the Pi? Are the test realy so ressource intensive?

AndreasHeine, 2020-06-23:
does they happen concurrently or in sequence?

swamper123, 2020-06-23:
What do you mean exactly?

AndreasHeine, 2020-06-23:
in the test there are serveral test with server and client instances running server and client on a low spec. pi its kind of heavy for him!

swamper123, 2020-06-23:
[image]

This is my last test run. Every dot seems to represent one test and they are running one by one (I except a setup, run, tear down, like pytest normally does).

swamper123, 2020-06-23:
But they are not running all concurrently then (I hope). O.o
In that case it would be explaining, that if he limb anywhere he is still running a server on localhost and crash other tests as well....but not exactly where.

AndreasHeine, 2020-06-23:
failing at running run-tests.sh ...
```
pi@raspberrypi:~/opcua-asyncio $ sudo sh run-tests.sh
run-tests.sh: 2: run-tests.sh: pytest: not found
```
but:
```
pi@raspberrypi:~/opcua-asyncio $ pip3 install -r dev_requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pytest in /home/pi/.local/lib/python3.7/site-packages (from -r dev_requirements.txt (line 1)) (5.4.3)
Requirement already satisfied: pytest-asyncio in /home/pi/.local/lib/python3.7/site-packages (from -r dev_requirements.txt (line 2)) (0.12.0)
Requirement already satisfied: coverage in /home/pi/.local/lib/python3.7/site-packages (from -r dev_requirements.txt (line 3)) (5.1)
Requirement already satisfied: pytest-cov in /home/pi/.local/lib/python3.7/site-packages (from -r dev_requirements.txt (line 4)) (2.10.0)
Requirement already satisfied: wcwidth in /usr/lib/python3/dist-packages (from pytest->-r dev_requirements.txt (line 1)) (0.1.7)
Requirement already satisfied: py>=1.5.0 in /home/pi/.local/lib/python3.7/site-packages (from pytest->-r dev_requirements.txt (line 1)) (1.8.2)
Requirement already satisfied: importlib-metadata>=0.12; python_version < "3.8" in /home/pi/.local/lib/python3.7/site-packages (from pytest->-r dev_requirements.txt (line 1)) (1.6.1)
Requirement already satisfied: attrs>=17.4.0 in /home/pi/.local/lib/python3.7/site-packages (from pytest->-r dev_requirements.txt (line 1)) (19.3.0)
Requirement already satisfied: pluggy<1.0,>=0.12 in /home/pi/.local/lib/python3.7/site-packages (from pytest->-r dev_requirements.txt (line 1)) (0.13.1)
Requirement already satisfied: packaging in /home/pi/.local/lib/python3.7/site-packages (from pytest->-r dev_requirements.txt (line 1)) (20.4)
Requirement already satisfied: more-itertools>=4.0.0 in /home/pi/.local/lib/python3.7/site-packages (from pytest->-r dev_requirements.txt (line 1)) (8.4.0)
Requirement already satisfied: zipp>=0.5 in /home/pi/.local/lib/python3.7/site-packages (from importlib-metadata>=0.12; python_version < "3.8"->pytest->-r dev_requirements.txt (line 1)) (3.1.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from packaging->pytest->-r dev_requirements.txt (line 1)) (1.12.0)
Requirement already satisfied: pyparsing>=2.0.2 in /usr/lib/python3/dist-packages (from packaging->pytest->-r dev_requirements.txt (line 1)) (2.2.0)
pi@raspberrypi:~/opcua-asyncio $
```
 maybe its not in path .... sorry i am not that fast on linux...

swamper123, 2020-06-23:
No worries. :)
You can run the test via: `./run-test.sh` in terminal. There is also a difference between sudo pip3 and pip3, because you are 2 different "users".
Try running `sudo pip3 install -r dev_requirements.txt` this shall install it for the sudo.

swamper123, 2020-06-23:
But please watch out, not that your pi starts heating up to much.

AndreasHeine, 2020-06-23:
if it starts to smell i will swith him off ;)

AndreasHeine, 2020-06-23:
looks like i stuc there:

```
pi@raspberrypi:~/opcua-asyncio $ sudo sh run-tests.sh
==================================================================================================================================================== test session starts ====================================================================
platform linux -- Python 3.7.3, pytest-5.4.3, py-1.8.2, pluggy-0.13.1
rootdir: /home/pi/opcua-asyncio, inifile: pytest.ini, testpaths: tests
plugins: cov-2.10.0, asyncio-0.12.0
collected 454 items

tests/test_client.py ........
tests/test_common.py .............................................................................................................................
tests/test_connections.py ..
tests/test_crypto_connect.py .....
```

swamper123, 2020-06-23:
I am starting to acceppt the faith that test are not running in a nice way. They may pass, stuc or need an eternity. I am not sure if a VM with Buster is a representitv solution for this, especially if the real hardware may behave another way.

So my VM is passing the tests now (and I will have a look if they pass in Python 3.8 as well), but I am not sure if we may delete the Raspberry Part out of readme, if we can't verify the tests without burning the hardware (or we need to add that a propper cooling system has to be provided, maybe it helps 😄 ).

Maybe with a RPi4 this problem could be under control, afaik it is trying to not overheat while running?

AndreasHeine, 2020-06-23:
3h later:
```
============================================================================================================ test session starts ============================================================================================================
platform linux -- Python 3.7.3, pytest-5.4.3, py-1.8.2, pluggy-0.13.1
rootdir: /home/pi/opcua-asyncio, inifile: pytest.ini, testpaths: tests
plugins: cov-2.10.0, asyncio-0.12.0
collected 454 items

tests/test_client.py ........                                                                                                                                                                                                         [  1%]
tests/test_common.py .............................................................................................................................                                                                                    [ 29%]
tests/test_connections.py ..                                                                                                                                                                                                          [ 29%]
tests/test_crypto_connect.py ........

```

swamper123, 2020-06-24:
I hope your Pi survived the night. 😄
But well I think this is my hardware problem. I owe you a beer @AndreasHeine for helping out. 
Cheers!

swamper123, 2020-06-24:
@oroulet Could you notice that in the readme, that running tests on a Raspberry may cause trouble/take an eternity?

oroulet, 2020-06-24:
Why does it take an eternity? That is the interesting question..  can we reproduce using 'cpulimit'?

swamper123, 2020-06-24:
It seems that the tests are to ressource intensive for the hardware. I don't know if the Pi Zero is reducing it's performance if it heats up, so this would explain why it may takes an eternity (but not start "melting" like my Pi3+).

swamper123, 2020-06-24:
[image]
I got the tipp to use ssh and request via `top`. You can see that the RAM was eaten alive while running this test. Other tasks seems to get blocked.
cpulimit will not help very much in that case, because Python would still use just one core.

swamper123, 2020-06-24:
So I left him running (with Fan and via ssh was still alive) and my Pi collapsed (has shutdowned himself). So either we have to optimise the tests for RAM usage or write a warning about running tests on the Py.

AndreasHeine, 2020-06-24:
have you tryed python 32bit / 64bit it makes a difference in ram usage!

swamper123, 2020-06-24:
I used the 32bit Raspbian OS for the pi, so there wasn't a 64bit version of Python. =/
But I saw that there is a 64bit Beta out for raspbian so I may give it a shot.

AndreasHeine, 2020-06-24:
64bit uses more ram than 32bit but if its 32bit ... :(

swamper123, 2020-06-24:
You can ssh on the Pi and then type `top` in the cmd. You will see a window like the one I posted. The process which is running will be pytest and you can see memory- and CPU-usage.

swamper123, 2020-06-24:
Or just run `top` parallel to your process, if you can...these 0,3% CPU usage wouldn't make any difference.

swamper123, 2020-06-26:
🎉 Well I got it running again without a collapse. 🎉 
What I have done:

- install all dependencies that are needed for Pythonmodules (like `libssl-dev` etc.)
- getting sources for Python3.8.3 and build Python with "--enable-optimizations" 
- clone a fresh repo
- setting up a better cooling system (seems to be not necessarry, but after the first blackout I wanted to be safe)
- run pytest
... and it passes. Everything, except the typical 5 skipped test. And this is not done in >3 hours, it's done in ~10min.

To be honest I'm not sure if this is caused because of the setted flag or about newest Python versions etc., but it runs like expected.

So maybe there is hope for your Pi Zero, too  @AndreasHeine ;)

closed 2020-06-26 by swamper123