FreeOpcUa/opcua-asyncio #224: OPC Foundation nodesets are no longer avalible ... !

AndreasHeine, 2020-06-24:
https://github.com/OPCFoundation/UA-Nodeset.git

AndreasHeine, 2020-06-24:
https://github.com/OPCFoundation/UA-Nodeset/issues/64 => please give it a thumb up!

> randy-armstrong can you please elaborate in more detail why the OPCF decided to remove the master branch and only make it available to OPC Foundation members?
> 
> In the OPC Day event today Stefan Hoppe mentioned multiple times that the O in OPC stands for openness.
> 
> Removing (hiding) the master branch behind a paywall is not really a sign for openness. This will also slow down the development of open source stacks, it almost seems as if commercial stack vendors are threatened by the good progress of the open source community and therefore asked to hide this content.
> 
> Can you please also clarify, preferably in the README:
> Will releases of the ua-nodeset still be available open source, or will the OPCF decide in the future to also remove these?

AndreasHeine, 2020-06-25:
https://github.com/FreeOpcUa/python-opcua/issues/1078

AndreasHeine, 2020-06-28:
there back avalible!

closed 2020-06-28 by AndreasHeine

oroulet, 2020-06-28:
great!

swamper123, 2020-06-29:
I am happy that they made it clear in their readme and made 1.04 default.