FreeOpcUa/opcua-asyncio #225: Error- Example - client_to_kepware.py

SAULSALBAL, 2020-06-26:
Hello, 

I am trying to read a tag from KEPServerEX 6 with the example - client_to_kepware, when I run the program I get the following error:

**Error:**
`C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:28: RuntimeWarning: coroutine 'Client.connect' was never awaited
  client.connect()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:31: RuntimeWarning: coroutine 'Node.get_children' was never awaited
  print("childs of root are: ", root.get_children())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:32: RuntimeWarning: coroutine 'Node.read_browse_name' was never awaited
  print("name of root is", root.read_browse_name())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:34: RuntimeWarning: coroutine 'Node.get_children' was never awaited
  print("childs og objects are: ", objects.get_children())
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:38: RuntimeWarning: coroutine 'Node.read_value' was never awaited
  print(f"tag1 is: {tag1} with value {tag1.read_value()} ")
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:40: RuntimeWarning: coroutine 'Node.read_value' was never awaited
  print(f"tag2 is: {tag2} with value {tag2.read_value()} ")
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py:55: RuntimeWarning: coroutine 'Client.disconnect' was never awaited
  client.disconnect()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Traceback (most recent call last):
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 44, in <module>
    handle1 = sub.subscribe_data_change(tag1)
AttributeError: 'coroutine' object has no attribute 'subscribe_data_change'
Root is i=84
childs of root are:  <coroutine object Node.get_children at 0x00000200456ED3C8>
name of root is <coroutine object Node.read_browse_name at 0x00000200456ED3C8>
childs og objects are:  <coroutine object Node.get_children at 0x00000200456ED3C8>
tag1 is: ns=2;s=Channel1.Device1.Tag1 with value <coroutine object Node.read_value at 0x00000200456ED3C8> 
tag2 is: ns=2;s=Channel1.Device1.Tag2 with value <coroutine object Node.read_value at 0x00000200456ED3C8> 
sys:1: RuntimeWarning: coroutine 'Client.create_subscription' was never awaited
`

**This is my url to connect to kepware:**
client = Client("opc.tcp://127.0.0.1:49320/OPCUA/SimulationServer/")

AndreasHeine, 2020-06-26:
looks like there are some missing "await" statements!

AndreasHeine, 2020-06-26:
```
import sys
sys.path.insert(0, "..")
import logging

from asyncua import Client

class SubHandler(object):

    """
    Client to subscription. It will receive events from server
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

if __name__ == "__main__":
    #from IPython import embed
    logging.basicConfig(level=logging.WARN)
    client = Client("opc.tcp://192.168.56.100:49320/OPCUA/SimulationServer/")
    #client = Client("opc.tcp://192.168.56.100:4840/OPCUA/SimulationServer/")
    #client = Client("opc.tcp://olivier:olivierpass@localhost:53530/OPCUA/SimulationServer/")
    try:
        await client.connect()
        root = client.nodes.root
        print("Root is", root)
        print("childs of root are: ", root.get_children())
        print("name of root is", root.read_browse_name())
        objects = client.nodes.objects
        print("childs og objects are: ", objects.get_children())

        tag1 = await client.get_node("ns=2;s=Channel1.Device1.Tag1")
        print(f"tag1 is: {tag1} with value {tag1.read_value()} ")
        tag2 = await client.get_node("ns=2;s=Channel1.Device1.Tag2")
        print(f"tag2 is: {tag2} with value {tag2.read_value()} ")

        handler = SubHandler()
        sub = await client.create_subscription(500, handler)
        handle1 = await sub.subscribe_data_change(tag1)
        handle2 = await sub.subscribe_data_change(tag2)

        from IPython import embed
        embed()

        
        await sub.unsubscribe(handle1)
        await sub.unsubscribe(handle2)
        await sub.delete()
    finally:
        await client.disconnect()
```

SAULSALBAL, 2020-06-29:
Hello,

I just tried updating the code and now I get this error:

`C:\Users\User\AppData\Local\Programs\Python\Python37\python.exe C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py
WARNING:asyncua.client.client:Requested session timeout to be 3600000ms, got 60000ms instead
WARNING:asyncua.client.ua_client.UASocketProtocol:ServiceFault from server received  in response to ActivateSessionRequest
Traceback (most recent call last):
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 44, in <module>
    asyncio.run(main())
  File "C:\Users\User\AppData\Local\Programs\Python\Python37\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\User\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 25, in main
    async with Client(url=url) as client:
  File "..\asyncua\client\client.py", line 71, in __aenter__
    await self.connect()
  File "..\asyncua\client\client.py", line 234, in connect
    await self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)
  File "..\asyncua\client\client.py", line 443, in activate_session
    return await self.uaclient.activate_session(params)
  File "..\asyncua\client\ua_client.py", line 302, in activate_session
    data = await self.protocol.send_request(request)
  File "..\asyncua\client\ua_client.py", line 144, in send_request
    self.check_answer(data, f" in response to {request.__class__.__name__}")
  File "..\asyncua\client\ua_client.py", line 153, in check_answer
    hdr.ServiceResult.check()
  File "..\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadIdentityTokenInvalid: "The user identity token is not valid."(BadIdentityTokenInvalid)`

**This is my URL:**
`url = "opc.tcp://localhost:49320/OPCUA/SimulationServer/"`

Do you know what the problem could be?

Thanks.

closed 2023-03-29 by schroeder-