FreeOpcUa/opcua-asyncio #227: The user identity token is not valid. Example - client_to_kepware.py

SAULSALBAL, 2020-06-29:
Hello,

I just tried updating the code and now I get this error:

`C:\Users\User\AppData\Local\Programs\Python\Python37\python.exe C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py
WARNING:asyncua.client.client:Requested session timeout to be 3600000ms, got 60000ms instead
WARNING:asyncua.client.ua_client.UASocketProtocol:ServiceFault from server received  in response to ActivateSessionRequest
Traceback (most recent call last):
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 44, in <module>
    asyncio.run(main())
  File "C:\Users\User\AppData\Local\Programs\Python\Python37\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\User\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 25, in main
    async with Client(url=url) as client:
  File "..\asyncua\client\client.py", line 71, in __aenter__
    await self.connect()
  File "..\asyncua\client\client.py", line 234, in connect
    await self.activate_session(username=self._username, password=self._password, certificate=self.user_certificate)
  File "..\asyncua\client\client.py", line 443, in activate_session
    return await self.uaclient.activate_session(params)
  File "..\asyncua\client\ua_client.py", line 302, in activate_session
    data = await self.protocol.send_request(request)
  File "..\asyncua\client\ua_client.py", line 144, in send_request
    self.check_answer(data, f" in response to {request.__class__.__name__}")
  File "..\asyncua\client\ua_client.py", line 153, in check_answer
    hdr.ServiceResult.check()
  File "..\asyncua\ua\uatypes.py", line 224, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadIdentityTokenInvalid: "The user identity token is not valid."(BadIdentityTokenInvalid)`

**This is my URL:**
`url = "opc.tcp://localhost:49320/OPCUA/SimulationServer/"`

Do you know what the problem could be?

Thanks.

_Originally posted by @SAULSALBAL in https://github.com/FreeOpcUa/opcua-asyncio/issues/225#issuecomment-651231037_

AndreasHeine, 2020-06-29:
serverconfiguration? security policie? your client code? we need more details!

SAULSALBAL, 2020-06-29:
Hello

Thank you for your answers.

**Server Configuration:**
[image]
:
**Security Policies and server configuration**
[image]

**Client-Code:**
`import sys
import asyncio
sys.path.insert(0, "..")
import logging

from asyncua import Client, ua

class SubHandler(object):

    """
    Subscription Handler. To receive events from server for a subscription
    """

    def datachange_notification(self, node, val, data):
        print("Python: New data change event", node, val)

    def event_notification(self, event):
        print("Python: New event", event)

async def main():
    url = "opc.tcp://localhost:49320/"
    async with Client(url=url) as client:
        print("Root children are", await client.nodes.root.get_children())

        tag1 = await client.get_node("ns=2;s=Channel1.Device1.Tag1")
        print(f"tag1 is: {tag1} with value {await tag1.read_value()} ")
        tag2 = await client.get_node("ns=2;s=Channel1.Device1.Tag2")
        print(f"tag2 is: {tag2} with value {await tag2.read_value()} ")

        handler = SubHandler()
        sub = await client.create_subscription(500, handler)
        handle1 = await sub.subscribe_data_change(tag1)
        handle2 = await sub.subscribe_data_change(tag2)

        # await sub.unsubscribe(handle1)
        # await sub.unsubscribe(handle2)
        # await sub.delete()

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    asyncio.run(main())`

AndreasHeine, 2020-06-29:
looks ok!

you should check with wireshark where the connect/activate session breaks

SAULSALBAL, 2020-06-29:
I was able to solve the above problem, but now I get this error.

**ERROR:**
`C:\Users\User\AppData\Local\Programs\Python\Python37\python.exe C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py
WARNING:asyncua.client.client:Requested session timeout to be 3600000ms, got 60000ms instead
Traceback (most recent call last):
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 44, in <module>
    asyncio.run(main())
  File "C:\Users\User\AppData\Local\Programs\Python\Python37\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\User\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "C:/Users/User/PycharmProjects/opcua-asyncio/examples/client_to_kepware.py", line 28, in main
    tag1 = await client.get_node("ns=2;s=Channel1.Device1.Tag1")
TypeError: object Node can't be used in 'await' expression
Root children are [Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=87))]`

AndreasHeine, 2020-06-29:
`tag1 = await client.get_node("ns=2;s=Channel1.Device1.Tag1")`

this should work:

`tag1 = client.get_node("ns=2;s=Channel1.Device1.Tag1")`

have not seen that the get_node method isn't a async def so you dont need to await it!
sorry about that!

SAULSALBAL, 2020-06-29:
Done, the problem was solved.

Thank you so much.

closed 2020-06-29 by SAULSALBAL