FreeOpcUa/opcua-asyncio #228: Make other simple DataType Nodes

swamper123, 2020-06-30:
Hello together,

so the most time I dealt with the typical build-in Datatypes like int64, float, DateTime etc. and know how to make an extended datatype.

While scrolling throw the Address Space Model (Part 3) in the norms, there are even DataTypes like DateString, DurationString, DecimalString, ImageGIF etc. etc. 

Is there already a way to create Nodes with such a DataType (and I am completly stumped) or is this a thing we shall deal with in future?
(Do we may need an example for such DataTypes?)

oroulet, 2020-06-30:
The UA datamodel is a bit cumbersome. you have the buildin (variant) types you can find in ua.VariantType.xxx like float, etc... this is are in reality the only "real" types. Then you can flag a node to be of whatever type you find under Root/DataTypes/  this is little more than documentation, then you have the the ua.VariantType.Extensionobjects which are structures and defined in a quite complicated way (I think there is an example to use them)

swamper123, 2020-06-30:
I totally agree with you that this overview charts look pretty easy, but in detail the modelling is a bit tuff. 😄 

I found something in the specs about that: _"An ExtensionObject is a container for any Structured DataTypes which cannot be encoded as one of the othe rbuilt-in data types._" - OPC Spec v1.04, Part 6- 5.1.5
So if I understood this right, going via ExtensionObject would only be the way for structured DataTypes.

Beside the structured Data Types exist _Simple DataTypes_ and _Enumeraiton DataTypes,_ (and abstract DataTyoes, but I let this one out for now) which would fell out of these ExtensionObjects:
- The Simple DataTypes are subtypes of the Built-in ones and are handled like that....but in the end they aren't a built-in one.
- Similar problem with the Enumeration ones, but I  skip that as well for this issue here.

 Just for the record for others: There are different Data Types:
- Built-in (like Double or Int32)
- Simple (Datestring, Duration ...)
- Structured (Like Argument, TimeZoneDataType...)
- Enumeration (I don't know a good example for that one right now)
- Abstract (but this one has no encoding, because well...it is Abstract. Like "Integer" (because there are many Subtypes of an Integer like Int64/32...)
(source; [opc-ua spec part 3, 5.8.2. ](https://reference.opcfoundation.org/v104/Core/docs/Part3/5.8.2/))

So I am still struggleing with_simple DataTypes_. =/

oroulet, 2020-06-30:
As far as I remember Duration is only a node data type. So you  can set the node datatype to be Duration while you set the value using VariantType.Float or DateTime