FreeOpcUa/opcua-asyncio #230: uatypes.py -> class DataValue

AndreasHeine, 2020-07-07:
https://github.com/FreeOpcUa/opcua-asyncio/blob/55d76f6748a6fdb3dbf05d68f1bf9633586b2688/asyncua/ua/uatypes.py#L838
```

    def __init__(self, variant=None, status=None):
        self.Encoding = 0
        if not isinstance(variant, Variant):
            variant = Variant(variant)
        self.Value = variant
        if status is None:
            self.StatusCode = StatusCode()
        else:
            self.StatusCode = status
        self.SourceTimestamp = None  # DateTime()
        self.SourcePicoseconds = None
        self.ServerTimestamp = None  # DateTime()
        self.ServerPicoseconds = None
        self._freeze = True
```

why is self.SourceTimestamp and self.ServerTimestamp = None? 
guess it should be like `datetime.utcnow()` were there issues in the past?

its in both python-opcua and asyncua!

swamper123, 2020-07-08:
[Some servers do not accept timestamps](https://github.com/FreeOpcUa/python-opcua/commit/0268d918143c90c64a09678560770be6184a91ca) was the reason use `None `instead `of datetime.now()` (this was used before, `datetime.utcnow()` which would had been the more correct way).

AndreasHeine, 2020-07-08:
> Aren't timestamps mandatory?
havn't seen anything about that...

i would change it like this to have all options open:
```python
    def __init__(self, variant=None, status=None, sourceTimestamp=None, sourcePicoseconds=None, serverTimestamp=None, serverPicoseconds=None):
        self.Encoding = 0
        if not isinstance(variant, Variant):
            variant = Variant(variant)
        self.Value = variant
        if status is None:
            self.StatusCode = StatusCode()
        else:
            self.StatusCode = status
        self.SourceTimestamp = sourceTimestamp
        self.SourcePicoseconds = sourcePicoseconds
        self.ServerTimestamp = serverTimestamp
        self.ServerPicoseconds = serverPicoseconds
        self._freeze = True
```

swamper123, 2020-07-08:
I haven't found a definite parameter for that, but it seems to be optional on both sides.
> i would change it like this to have all options open:

I would approve this change.

closed 2023-03-29 by schroeder-