FreeOpcUa/opcua-asyncio #232: Unable to set custom url as endpoint

padraic-padraic, 2020-07-09:
Hello,

I am trying to setup an OPC-UA server inside a Docker container (based on debian-buster) and I'm not quite clear on the behaviour of `Server.set_endpoint`. If I try to use anything other than `0.0.0.0`, I get an error from asyncio that it cannot bind to the address. For example, using localhost gives

```
server.set_endpoint(`opc.tcp://localhost:4840`) .....
OSError: [Errno 99] error while attempting to bind on address ('::1', 4840, 0, 0): cannot assign requested address
```

And a similar error, with the public IPv4 addres substituted for `::1`, if using the server IP.

Am I not using this correctly?

If I do use 0.0.0.0 then the resulting endpoints have address like 172.x.x.x with a discovery URL of 0.0.0.0

ZuZuD, 2020-07-09:
Better listening on 0.0.0.0 from docker and bind the port from the host, example ```docker run -p 8080:4840 docker_image```. Also, this is article is worth 5min of your time https://pythonspeed.com/articles/docker-connection-refused/

ondrejnovakcvut, 2021-08-31:
When binding to 0.0.0.0 in the Docker container, the server will listen on 172.x.x.x and the discovery will return container IP address (172.x.x.x), not the host IP (physical one). I think the IP address is dynamically rewritten in `InternalServer.get_endpoints()` to match the client network. In case of Docker, it's the IP in range 172.x.x.x.

If it would be possible to disable this automatic translation (either globally or for specific endpoints), then by adding an endpoint with a custom IP by calling `InternalServer.add_server()`, an working discovery could be deployed in Docker.

kohtala, 2021-10-05:
Hi. The docker is one where I too am facing this, but Kubernetes ingress and load balancers are others. Any deployment that has the OPC-UA server is behind a proxy/NAT/load balancer that offers client a different IP address than where the OPC-UA server is implemented needs some solution.

To clarify for those not yet familiar with Docker and containers, the container runtime usually starts the process in a network namespace that has own network interface and IP address for the process. (Container also has own mount, hostname, IPC, PID namespaces, for PID 1 in the container, and sometimes also user ID namespace.) The network interface is connected to a software switch/bridge and iptables is configured to DNAT towards the containers from ports on the host IP address and masquerade for outbound connections.

This means the OPC-UA server network namespace does not have the client facing public IP and therefore it can not listen on it. The `Server.set_endpoint` is usually called with an URL with netloc 0.0.0.0:4840.

With the client, the sequence of events is that
1. the client connects to the DNAT exposed port on the host IP address,
2. the client sends `GetEndpointsRequest` and `CreateSessionRequest`,
3. the destination address gets converted by DNAT to the server interface address in the container,
4. the `UaProcessor` reads the `sockname` from the `Transport`, which is now the server address in the container,
5. the server patches it into the endpoint netloc it returns for `GetEndpointsRequest` and `CreateSessionRequest`
6. the client tries to connect to an endpoints it can not connect to.

This would have some more subtle cases where some clients are in front and other behind the load balancer (other containers on the same bridge). In some cases those behind the load balancer may be firewalled from connecting to the front IP address (network policies in Kubernetes).

In some extreme case beyond containers, some load balancers support own protocol where they pass the client IP address to the server. Like HAProxy proxy protocol. I believe it would be another transport. The `Server` and `BinaryServer` are difficult to replace for special application needs because they implement some other functionality.

I believe all the problem could be solved if the application could provide a replacement server factory for the `loop.create_server` in `BinaryServer.start`. I'm not sure how difficult it would be to implement an own server that would offer a custom transport only for the modified `sockname` for the public IP address, so another setting to distinguish the server public endpoint from the listen endpoint could be helpful.

While waiting for an asyncua API that support this use case, I've patched the iserver.getendpoints:

        server = Server()
        await server.init()

        parsed_url = urlparse(url)
        listenurl = parsed_url._replace(netloc="0.0.0.0:4840").geturl()
        server.set_endpoint(listenurl)
        await server.set_application_uri(url)
        parsed_url = urllib.parse.urlparse(url)
        orig_get_endpoints = server.iserver.get_endpoints

        def get_endpoints(params=None, sockname=None):
            sockaddr = parsed_url.netloc.split(':')
            if len(sockaddr) == 1:
                sockaddr.append(4840)
            return orig_get_endpoints(params=params, sockname=sockaddr)
        server.iserver.get_endpoints = get_endpoints

Ugly.