FreeOpcUa/opcua-asyncio #233: Browsing HasSubType references without "includeSubtypes"

genuaboro, 2020-07-13:
Hi,

while browsing nodes in the example OPC UA server (examples/server-example.py) I noticed something weird (at least weird to me).

If I specify the HasSubType reference for the "referenceTypeId" parameter in the browse request, I will only get these references in the browse response if I also set the "includeSubtypes" parameter to true.
Other reference type IDs in the browse request wont show this behavior and I will get the requested references without setting includeSubTypes.

I only took a brief look at the code, but I would guess the behavior comes from 5f488ab2 (not verified, that this is the code path taken):

```python
    def _suitable_reftype(self, ref1, ref2, subtypes):
        """
        """
        if not subtypes and ref2.Identifier == ua.ObjectIds.HasSubtype:
            return False
```

I didn't see anything in the OPC UA specification that would account for special handling of the HasSubType reference type in this case. Other OPC UA stacks also return references with HasSubType directly (without setting "IncludeSubtypes").

Is there a reason why the OPC UA server behaves this way?

Best regards

genuaboro, 2020-07-21:
I wrote some python code to illustrate the issue. (I'm not a python programmer, so apologies for any style issues etc.)

```python
import asyncio
import logging

from asyncua import ua
from asyncua import Client

logging.basicConfig(level=logging.ERROR)
_logger = logging.getLogger('asyncua')

async def main():
    url = "opc.tcp://localhost:4840"
    async with Client(url=url) as client:
        params = ua.BrowseParameters()

        # all references for node "BASEOBJECTTYPE" (ns=0;i=58)
        desc = ua.BrowseDescription()
        desc.NodeId = ua.NodeId(58, 0)
        desc.BrowseDirection = ua.BrowseDirection.Both
        desc.ResultMask = ua.BrowseResultMask.All
        params.NodesToBrowse.append(desc)

        # references with type "Organizes" (no subtypes)
        desc = ua.BrowseDescription()
        desc.NodeId = ua.NodeId(58, 0)
        desc.BrowseDirection = ua.BrowseDirection.Both
        desc.ReferenceTypeId = ua.NodeId(35, 0)
        desc.IncludeSubtypes = False
        params.NodesToBrowse.append(desc)

        # references with type "Organizes" (including subtypes)
        desc = ua.BrowseDescription()
        desc.NodeId = ua.NodeId(58, 0)
        desc.BrowseDirection = ua.BrowseDirection.Both
        desc.ReferenceTypeId = ua.NodeId(35, 0)
        desc.IncludeSubtypes = True
        params.NodesToBrowse.append(desc)

        # references with type "HasSubType" (no subtypes)
        desc = ua.BrowseDescription()
        desc.NodeId = ua.NodeId(58, 0)
        desc.BrowseDirection = ua.BrowseDirection.Both
        desc.ReferenceTypeId = ua.NodeId(45, 0)
        desc.IncludeSubtypes = False
        params.NodesToBrowse.append(desc)

        # references with type "HasSubType" (including subtypes)
        desc = ua.BrowseDescription()
        desc.NodeId = ua.NodeId(58, 0)
        desc.BrowseDirection = ua.BrowseDirection.Both
        desc.ReferenceTypeId = ua.NodeId(45, 0)
        desc.IncludeSubtypes = True
        params.NodesToBrowse.append(desc)

        results = await client.uaclient.browse(params)

        _logger.error("Count of all references:\t\t\t\t%d",
                      len(results[0].References))
        _logger.error("Count of 'Organizes' references (no subtype):\t%d",
                      len(results[1].References))
        _logger.error("Count of 'Organizes' references (with subtypes):\t%d",
                      len(results[2].References))
        _logger.error("Count of 'HasSubType' references (no subtype):\t%d",
                      len(results[3].References))
        _logger.error("Count of 'HasSubType' references (with subtype):\t%d",
                      len(results[4].References))

        # to prove that we did not get real sub types in the resulting
        # references, we manually count the 'HasSubType' references here
        count_subtype = 0
        count_all = 0
        for r in results[0].References:
            count_all += 1
            if r.ReferenceTypeId == ua.NodeId(45, 0):
                count_subtype += 1

        _logger.error("Manual count of 'HasSubType' references:\t\t%d", count_subtype)
        _logger.error("Manual count of all references:\t\t\t%d", count_all)

if __name__ == "__main__":
    asyncio.run(main())
```

If I run this code against the minimal example server (examples/server-minimal.py), I get the following output:
> ERROR:asyncua:Count of all references:                          30
> ERROR:asyncua:Count of 'Organizes' references (no subtype):     1
> ERROR:asyncua:Count of 'Organizes' references (with subtypes):  1
> ERROR:asyncua:Count of 'HasSubType' references (no subtype):    0
> ERROR:asyncua:Count of 'HasSubType' references (with subtype):  29
> ERROR:asyncua:Manual count of 'HasSubType' references:          29
> ERROR:asyncua:Manual count of all references:                   30

The server-example.py produces similar output (1 additional HasSubType reference).

If I run this code against a minimal open62541 server, I get the following output:

> ERROR:asyncua:Count of all references:                          13
> ERROR:asyncua:Count of 'Organizes' references (no subtype):     1
> ERROR:asyncua:Count of 'Organizes' references (with subtypes):  1
> ERROR:asyncua:Count of 'HasSubType' references (no subtype):    10
> ERROR:asyncua:Count of 'HasSubType' references (with subtype):  10
> ERROR:asyncua:Manual count of 'HasSubType' references:          10
> ERROR:asyncua:Manual count of all references:                   13

The difference is in the 4th result, where the asyncio server doesn't return any references for the browse description.

I hope this illustrates, that you don't get HasSubType references from an asyncio OPC UA server unless you set the IncludeSubtypes parameter to true.

AndreasHeine, 2020-07-21:
just for my general understanding of your problem:

> ERROR:asyncua:Count of 'HasSubType' references (no subtype): 0

so you would expect that if you use `IncludeSubtypes = False` you still get the subtypes?

genuaboro, 2020-07-21:
> so you would expect that if you use `IncludeSubtypes = False` you still get the subtypes?

"includeSubtypes" relates to the referenceTypeId in the browse description, not the nodeId that is browsed.

From the specification Part 4 5.8.2:
> includeSubtypes (Boolean): Indicates whether subtypes of the ReferenceType should be included in the browse. If TRUE, then instances of referenceTypeId and all of its subtypes are returned.

If I set the parameter to false, I should only get references with the specified referenceTypeId. If the parameter is true, I should also get references with a reference type that is a sub type of the referenceTypeId.

There is no actual special case for 'HasSubType' references (although the name and the parameter might by similar and confusing in this case).

If I have set the referenceTypeId to "HasSubType", I should always get references with this reference type - independent of the "includeSubtypes" parameter. If I set the "includeSubtypes" parameter to true, I should additionally get references where the reference type is a sub type of "HasSubType"

This is at least my understanding of the specification and what I witnessed in the behavior of other OPC UA stacks.

oroulet, 2020-08-13:
@genuaboro I did not follow closely the description but you are probably right. fix welcome if you have some ideas about how it should behaves

PR #1018 (2022-09-07): Fix interpretation of IncludeSubtypes
Fix interpretation of IncludeSubtypes in Browse requests.

According to the spec, IncludeSubtypes indicates whether subtypes of the ReferenceType should be included in the browse. If TRUE, then instances of ReferenceType and its subtypes are returned.

The current implementation has a different interpretation: Get instances of ReferenceType and subtypes thereof. If IncludeSubtypes is FALSE, then exclude the type "HasSubtype".

The same applies to RelativePath requests.

Fixes #233.
There may be conflicts with #1017, I can rebase when it got merged.

**WARN [... cut]
closed 2022-09-07 by oroulet