FreeOpcUa/opcua-asyncio #234: [Topic] Support of DataTypeDefinition

driftregion, 2020-07-14:
Hi,
I'm using `python-opcua` to interact with a couple of Siemens PLCs: an S7-1511 and an S7-1212.  I'm finding that their implementations of user-defined types are different.

## Ways to access user-defined type information
| S7-1511 | S7-1212 |
| - | - |
| `TypeDictionary`, `DataTypeDefinition` | `DataTypeDefinition` |

Apparently, the `TypeDictionary` approach has been removed in OPC-UA 1.04.  This post has more information:

https://forum.prosysopc.com/forum/opc-ua-java-sdk/get-datatype-from-siemens-s7-1500-server-to-add-to-typedictionary/#p4112 

Revision 1.04 Highlights (copied from OPC-10000-3: Address Space Model, Release 1.04)
| Mantis ID | Summary | Resolution |
| - | - | - |
| 3018 | Handling of DataType Encoding Information | Added Property to DataType NodeClass in 5.8.3 containing information about data type definition. Added DataTypes for handling the information in 8.48, 8.49, 8.50, 8.51, and 8.52. Removed the old approach having DataTypeDictionaries. This effects 5.6.2, where Properties have been removed, 5.8, where the old approach was defined in detail, and 5.5.1 as well as 7, where the ReferenceType HasDefinition and its usage was removed. The old approach is moved to an annex of Part 5 and can still be applied by OPC UA Applications. |

I'm able to read the `DataTypeDefinition` as follows, but haven't yet figured out how to parse it.
```python
type_node = self.client.get_node(variable_node.get_data_type())
dv = type_node.get_attribute(opcua.ua.attribute_ids.AttributeIds.DataTypeDefinition)
```

oroulet, 2020-07-14:
OK looks like we have some work to do. 
Can you attach content of that attribute? is it xml? or maybe base64 encoded data?

oroulet, 2020-07-14:
that sounds anyway like a better solution thatn the TypeDictonnary stuff...

driftregion, 2020-07-15:
It's either a `StructureDefinition` or an `EnumDefinition` defined in `OPC 10000-3: Address Space Model`.
 
[image]

I tried decoding a `StructureDefinition` as follows the other day, but ran into a de-serialization issue, perhaps self-inflicted.  Details here: https://support.industry.siemens.com/tf//ww/en/posts/failing-to-deserialize-structuredefinition-from-opc-ua-server/240544/?page=0&pageSize=10 
```python
type_node = self.client.get_node(variable_node.get_data_type())
import opcua.ua.attribute_ids
import opcua.ua.uaprotocol_auto
import opcua.ua.ua_binary
import opcua.common.utils
dv = type_node.get_attribute(opcua.ua.attribute_ids.AttributeIds.DataTypeDefinition)   
buf = opcua.common.utils.Buffer(dv.Value.Value.Body) 
opcua.ua.ua_binary.from_binary(opcua.ua.uaprotocol_auto.StructureDefinition, buf) 
```

swamper123, 2020-07-15:
I am not sure if you can get Attributes of an abstract class itself.
Like you said, it depends if it is a `StructureDefinition `or `EnumDefinition`. 

[image]

`StructureDefinition` again is:

[image]

Where the StructureField Structure is the better known:

[image]

Just for a better overview.

oroulet, 2020-07-19:
@driftregion I added code to parse the DataTypeDefinion and read it simply with Node.get_data_type_definition()
This is the part of support. Not sure how to continue..
Should we parse all nodes under BaseDataType/Structures and generate the code for them? probably...

oroulet, 2020-07-19:
Ok here is plan:
1) generate Class code from Structure de finition data. Hopefully I can copy most code from existing structure code
2) add Client.load_data_tyoe_definition() that check all nodes under baseDataTyoe/Structure and generate/load code for all unknown types
3) maybe add Node.Load_data_type_definition() so this can be done only for one node

Comments on API?

oroulet, 2020-07-19:
@driftregion can you test that branch? https://github.com/FreeOpcUa/opcua-asyncio/pull/238
I have no server to test so I am not sure it works...

oroulet, 2020-07-19:
And if someone has a link to a server on the internet Iimplementing that stuff it would be useful. Thanks

oroulet, 2020-07-20:
Works now

driftregion, 2020-07-23:
@oroulet, I just tried this out.  

sync
```python
#!/usr/bin/env python3
import opcua
import opcua.ua.attribute_ids
import opcua.ua.uaprotocol_auto
import opcua.ua.ua_binary
import opcua.common.utils

with opcua.Client("opc.tcp://192.168.1.3:4840") as client:
    var = client.get_node("ns=5;i=5")
    dt_id = var.get_data_type()
    print(f"located datatype {dt_id} of node at {var}")
    dt = client.get_node(dt_id)

    dv = dt.get_attribute(opcua.ua.attribute_ids.AttributeIds.DataTypeDefinition)   
    buf = opcua.common.utils.Buffer(dv.Value.Value.Body) 
    opcua.ua.ua_binary.from_binary(opcua.ua.uaprotocol_auto.StructureDefinition, buf)
```
raises
```sh
opcua.common.utils.NotEnoughData: Not enough data left in buffer, request for 4, we have Buffer(size:1, data:b'\x00')
```

async, using https://github.com/FreeOpcUa/opcua-asyncio/pull/238
```python
#!/usr/bin/env python3
import asyncio
import asyncua

async def main():
    async with asyncua.Client("opc.tcp://192.168.1.3:4840") as client:
        var = client.get_node("ns=5;i=5")
        dt_id = await var.read_data_type()
        print(f"located datatype {dt_id} of node at {var}")
        dt = client.get_node(dt_id)
        dtd = await dt.read_data_type_definition()
asyncio.run(main())
```
raises
```
asyncua.common.utils.NotEnoughData: Not enough data left in buffer, request for 4, we have 1
```

It's still unclear whether this is Siemens' fault or our fault.  In any case, UAExpert is able to decode the `StructureDefinition`

The [hexdump that I posted on the Siemens forums](https://support.industry.siemens.com/tf//ww/en/posts/failing-to-deserialize-structuredefinition-from-opc-ua-server/240544/?page=0&pageSize=10#pagetop) can be used to figure this out.  It is the binary payload returned by `await self.read_attribute(ua.AttributeIds.DataTypeDefinition)`

oroulet, 2020-07-23:
strange error. I tested it on a few servers and the parsing worked fine.
But I see that you example uses optional members. I probably did not meet a server with such structures yet. Might be a typo somewhere 

Can you paste the entire stack trace here?
and also can you paste a wireshark log of your async exampl above? I want to see what the server sent you

btw if you do not set your variable as optional it will probably work

oroulet, 2020-07-23:
just parsed the entire addresse space of prosys opcua server and I did not get any error. This is strnage, must be some optional fields there

EvashuangshuangFeng, 2022-05-16:
@oroulet  Hello, I have the same problem using nested structure. Has anyone fixed it? 

File "C:\Program Files\Python38\lib\site-packages\asyncua\common\utils.py", line 59, in read
    raise NotEnoughData(f"Not enough data left in buffer, request for {size}, we have {self._size}")
asyncua.common.utils.NotEnoughData: Not enough data left in buffer, request for 256, we have 95

EvashuangshuangFeng, 2022-05-16:
@driftregion Hello, Have you solved it？is there any good workaround to share about this issue?

oroulet, 2022-05-18:
As written, I could not find a server having the issue, so we cannot fix it. if you can expose a failing server on the internet we can try to have a look

driftregion, 2022-05-23:
@oroulet , It's possible that the Siemens OPC-UA server had a bug and that the response was invalid. That project has long since finished and I can't get a definitive conclusion now. Please see the post on the Siemens forums here:

https://support.industry.siemens.com/forum/ww/en/posts/failing-to-deserialize-structuredefinition-from-opc-ua-server/240544/?page=0&pageSize=10#pagetop

@EvashuangshuangFeng please open a new issue and include a wireshark log.

closed 2022-05-23 by driftregion