FreeOpcUa/opcua-asyncio #235: How can I set value to a Server Node using only NodeName from Client

Jayant-Pancholi, 2020-07-14:


AndreasHeine, 2020-07-14:
DisplayName?

Jayant-Pancholi, 2020-07-14:
My requirement is 
-> to connect a to server
-> Read all nodes and Objects
-> Set value to them using names (eg - PUMP_Pressure, 35)

AndreasHeine, 2020-07-14:
> -> Set value to them using names (eg - PUMP_Pressure, 35)

and again what do you mean with "name"!!!
DisplayName?
BrowseName?
Identifier?
NodeId?

Jayant-Pancholi, 2020-07-14:
using string identifier, DisplayName.

PUMP_RPM here ->

[image]

AndreasHeine, 2020-07-14:
this "NS2|Numeric|5" is identifier type "numeric"! you cant access it with string...

`ns=2;s=MyTemperature`
if you want to access a value with identifiertype string (like above) it has to be implemented serverside as identifiertype string and not numeric

Jayant-Pancholi, 2020-07-15:
How can I change the identifier type to String ?

AndreasHeine, 2020-07-15:
ua.NodeId.from_string()

AndreasHeine, 2020-07-15:
for i in range(100):
    var = vars_obj.add_variable(ua.NodeId.from_string(f'ns={address_space};s=DB_DATA.Test.var{i}'), f'Test.var{i}', 0)
    var.set_writable(True)
    var_list.append(var)

Jayant-Pancholi, 2020-07-15:
I fixed that part .

Thanks for your quick responses Andreas.

I am facing another error while running my async functions .

ValueError: The future belongs to a different loop than the one specified as the loop argument

How do i get this loop thing sorted.

AndreasHeine, 2020-07-15:
Cant help you without more information about your code... sound like you work with more then one eventloop!?

Jayant-Pancholi, 2020-07-15:
I have created one function as :-

async def ConnectToServerAsync(IP, Port):
    
    url = 'opc.tcp://' + str(IP) + ":" + str(Port) + '/freeopcua/server/'

    uri = 'http://examples.freeopcua.github.io'
    idx = await client.get_namespace_index(uri)
    async with Client(url=url) as client:
        print("Client LOOp", client.loop.is_running())
    while True:
        await asyncio.sleep(1)

and there is another function readOnly where I want to access Client nodes so that I can perform read write

async def ReadValueAsync(NodeName, VarName):
    var = client.get_node(ua.NumericNodeId(19, 2))
    var.read_value()
    x = await var.read_value()
    print(var)

AndreasHeine, 2020-07-15:
```python
import asyncio
from asyncua import Client, ua, Node

class SubscriptionHandler:
    """
    The SubscriptionHandler is used to handle the data that is received for the subscription.
    """
    def datachange_notification(self, node: Node, val, data):
        """
        Callback for asyncua Subscription.
        This method will be called when the Client received a data change message from the Server.
        """
        print('datachange_notification %r %s', node, val)

async def main():
    client = Client(url='opc.tcp://127.0.0.1:4840/test')
    async with client:
        idx = await client.get_namespace_index("http://andreas-heine.net/UA")
        handler = SubscriptionHandler()
        subscription = await client.create_subscription(50, handler)
        myobj = client.get_node("ns=2;i=1")
        print(myobj)
        nodes = await myobj.get_variables()
        await subscription.subscribe_data_change(nodes)

        while 1:
            await asyncio.sleep(0)

if __name__ == "__main__":
    asyncio.run(main())
```

maybe this structure helps you!

Jayant-Pancholi, 2020-07-16:
I have modified this structure -
The thing is like...Main is running altogether on a different thread itself.
Now I have created one more async function in which I am accessing the global object of client class using which I want to read write values to some nodes using its NS and I number.
client.

var = client.get_node(ua.NumericNodeId(19, 2))
    var.read_value()
    x = await var.read_value()
    print(var)

AndreasHeine, 2020-07-16:
it sounds like a bad idea to mix threads with asyncio... it makes it realy complex... maybe you can define the loop globaly (loop = asyncio.get_event_loop()) and put a threadlock around it!? and replace asyncio.run with loop.run_forever() !? appending each task to the loop!? 

this is more a python topic then a opcua one...

Jayant-Pancholi, 2020-07-16:
value of VAR is -> <coroutine object Node.read_value at 0x03CAD528>

How can I access the value of this node using VAR.

await VAR.read_value()

gives exception

AndreasHeine, 2020-07-16:
> gives exception

which one?

swamper123, 2020-07-16:
I can only confirm Andreas about mixing threads and asyncio. There are some methods to be threadsafe, but I don't understand why you have two eventloops anyway. As well, if you want real concourrency, you should be using multiple processes, instead of threads in python.

Could you paste the traceback? Sometimes `import tracemalloc` with `tracemalloc.start()` helps to find the exact position where an error may appear.

Jayant-Pancholi, 2020-07-16:
Traceback (most recent call last):
  File "D:/WORKSPACE/OPC/AsyncUA_OPCua_AsyncIO/CLIENT_GHB.py", line 285, in <module>
    p = ReadValue("PUMP4", "PUMP_TEMP")
  File "D:/WORKSPACE/OPC/AsyncUA_OPCua_AsyncIO/CLIENT_GHB.py", line 177, in ReadValue
    j = asyncio.run(ReadValueAsync(NodeName, VarName))
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "D:/WORKSPACE/OPC/AsyncUA_OPCua_AsyncIO/CLIENT_GHB.py", line 149, in ReadValueAsync
    await var.read_value()
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\common\node.py", line 165, in read_value
    result = await self.read_data_value()
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\common\node.py", line 176, in read_data_value
    return await self.read_attribute(ua.AttributeIds.Value)
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\common\node.py", line 280, in read_attribute
    result = await self.server.read(params)
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\client\ua_client.py", line 353, in read
    data = await self.protocol.send_request(request)
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\client\ua_client.py", line 140, in send_request
    data = await asyncio.wait_for(
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\asyncio\tasks.py", line 470, in wait_for
    fut = ensure_future(fut, loop=loop)
  File "C:\Users\jayant.pancholi\AppData\Local\Programs\Python\Python38-32\lib\asyncio\tasks.py", line 667, in ensure_future
    raise ValueError('The future belongs to a different loop than '
ValueError: The future belongs to a different loop than the one specified as the loop argument

swamper123, 2020-07-16:
Since you are using Python 3.8, you may choose `asyncio.create_task` instead of `ensure_future()`.

`ValueError: The future belongs to a different loop than the one specified as the loop argument`  -  like @AndreasHeine said, this is a problem of your project, not of this repo. Future objects are not thread safe, multithreadding and asyncio is a bad combo (but handable) and in the end using two diffrent eventloops in two threads is a bad combo.

Jayant-Pancholi, 2020-07-16:
Below is the complete code snippet of what I am trying .

The Intention is to Start a client and let it keep running continuously.
And a Separate Read/Write functions are made available to update the Node Objects and Variables on demand by any test application.
So a test application can call the StartClient and ReadValueAsync to use it effectively.

```
import asyncio
import pprint
import sys
import threading
import tracemalloc
from datetime import time

from asyncua.ua import NumericNodeId

sys.path.insert(0, "..")
import logging
from asyncua import Client, Node, ua

client = Client  #Creating Global Object of the OPC UA Client Class

async def ConnectToServerAsync(IP, Port):    
    global client    
    url = 'opc.tcp://' + str(IP) + ":" + str(Port) + '/freeopcua/server/'    
	
    async with Client(url=url) as client:
		print("Client LOOp", client.loop.is_running())		
        uri = 'http://examples.freeopcua.github.io'
        idx = await client.get_namespace_index(uri)

        #Reading all Entire Server namespace including Nodes, IDs, Objects, Methods, Variables and Attributes,
		# Storing them in local Data Structure (ObjectDict)

        while True:
            await asyncio.sleep(1)
 
async def ReadValueAsync(NS, Index):
    global client
    print("Client LOOP in READ", client.loop.is_running())
    var = await (client.get_node(ua.NumericNodeId(NS, Index))).read_value()
    return var

def StartAsyncClient(IP, Port):
    asyncio.run(ConnectToServerAsync(IP, Port))

def StartClient(IP, Port):
    x = threading.Thread(target=StartAsyncClient, args=(IP, Port))
    x.start()
    import time
    time.sleep(5)

if __name__ == '__main__':
    StartClient("192.168.43.36", 4843)
    p = asyncio.run(ReadValueAsync(19, 2)) # 19 as Index, Namespace = 2
    print("PVAL",p)
```

Getting Below error while reading -
    raise ValueError('loop argument must agree with Future')
ValueError: loop argument must agree with Future

AndreasHeine, 2020-07-16:
```python
import asyncio
from asyncua import Client

client = Client(url="opc.tcp://127.0.0.1:4840")

async def init(client):
    await client.connect()

async def task1(client):
    await asyncio.sleep(2)
    while 1:
        print("task1")
        print(await client.get_node("ns=0;i=2267").get_value())
        await asyncio.sleep(2)

async def task2(client):
    await asyncio.sleep(2)
    while 1:
        print("task2")
        print(await client.get_node("ns=2;s=DB_DATA.Test.var0").get_value())
        await asyncio.sleep(5)

if __name__ == '__main__':
    asyncio.ensure_future(init(client))
    asyncio.ensure_future(task1(client))
    asyncio.ensure_future(task2(client))
    asyncio.get_event_loop().run_forever()
```

result:

> task1
> task2
> 250
> 69
> task1
> 250
> task1
> 250
> task2
> 22
> task1
> 250
> task1
> 250

closed 2023-03-29 by schroeder-