FreeOpcUa/opcua-asyncio #241: UAClient transport should not be set to None silently on disconnect

HarrySky, 2020-07-20:
When disconnect happen `UAClient.transport` is silently set to `None` and I get `AttributeError`s instead of something more appropriate like `IOError`:

```python
AttributeError: 'NoneType' object has no attribute 'write'
```

I need to check if I am connected like that: `client.uaclient.protocol.transport is not None`.
Suggestion: throw `IOError` when trying to use transport with broken connection.

HarrySky, 2020-07-20:
Line in `UASocketProtocol._send_request` that tries to call `transport.write`:

https://github.com/FreeOpcUa/opcua-asyncio/blob/8b9670a5dd9d034c2e4eed8d599156201ad468d6/asyncua/client/ua_client.py#L130

HarrySky, 2020-07-20:
It seems that in sync version of this library you did not set `transport` (was called `_socket`) to `None`, but instead called `_socket.socket.close()`.

https://github.com/FreeOpcUa/python-opcua/blob/aed7ab0bbc8935a396b734634e378926a8dcbdf7/opcua/client/ua_client.py#L173

HarrySky, 2020-07-20:
Just checked what happens if it is not set to None. I will get this (using `uvloop`):

```python
RuntimeError: unable to perform operation on <TCPTransport closed=True reading=False 0x294fe68>; the handler is closed
```

Solution may be to check if `self.transport` exist before calling `write` (like it is checked in `UASocketProtocol.disconnect_socket` method). If it does not exist - raise `IOError`. Can create a PR for that.

But it will probably be a **breaking change**, since right now checking for existence of transport is the way to determine if disconnect happened and people might be doing this check not expecting `IOError`.

swamper123, 2020-07-20:
If I understand the code the right way:

https://github.com/FreeOpcUa/opcua-asyncio/blob/8b9670a5dd9d034c2e4eed8d599156201ad468d6/asyncua/client/ua_client.py#L47

[Like the docs say, it can be an exception object or None, if `connection_lost()` was called](https://docs.python.org/3.5/library/asyncio-protocol.html#connection-callbacks).

Is called, if the socket closes it's connection (and we overwrite in our class these methods). So there is a logger message, `self.transport` and the `self.state`, which can be checked before the write command and then throw an error or handleing it.
I haven't found a way in the docs, if there is a "standard way" to check the connection, but I would recommened to add a try/except block here, is `self.state ` is OPEN, before the write execution or even better checking it here and handle such things:
https://github.com/FreeOpcUa/opcua-asyncio/blob/8b9670a5dd9d034c2e4eed8d599156201ad468d6/asyncua/client/ua_client.py#L133

Maybe  `ConnectionError` could be the best description instead of `IOError`.

PR #248 (2020-07-25): Raise ConnectionError in send_request if state is not OPEN
Fixes #241 . Simple `try/catch` block that will raise `ConnectionError` in `send_request` if connection was lost
closed 2020-07-25 by oroulet