FreeOpcUa/opcua-asyncio #243: Extend Travis with higher Python versions?

swamper123, 2020-07-20:
Since we provide support for >=Python3.6, we may add Python3.7 and Python3.8 (actual stable) as well in out Travis tests?

AndreasHeine, 2020-07-21:
since 3.7 i lost track of the changes, are there fundamental changes in 3.8?

swamper123, 2020-07-21:
What may be usefull for us or could affect us:

In Python3.7:
- the `asyncio.run()` is introduced, which is already used in a tests as well
- some improvements for Windows devices
- asyncio.Lock is deprecated
- massive asyncio performance boost (get_event_loop, gather, sleep, overhead in debug mode)
- performance boosts for Python in general

In Python3.8:
- fewer swallowed exceptions (seems to happening in some cases)
- asyncio.CancelledError inherits from BaseException rather than Exception and no longer inherits from concurrent.futures.CancelledError (we already deal sometimes with newer Versions, like in test_crypto_connect.py)
- asyncio.wait_for() waits now for cancellation, if an asyncio.Task is used.
- walrus operator can be used 
- [Positional-only parameters](https://www.python.org/dev/peps/pep-0570/#how-to-teach-this)
- importlib.metadata provides metadata from third-party packages
- asyncio improvements for Windows
- IPv6 support for loop.create_connection()
- Pathlib returns boolean expressions for is_file(), exists() and more, instead of ValueError
- loop parameter is deprecated in many asyncio methodsand will be removed in later versions
- nicer `=` specifier for f-strings, so you can be more explicit how the variable shall be printed

I am using Python3.8 since April and hadn't got any big issues so far. Biggest break has been the replaced Exceptions, which we seem to handle in some cases.

oroulet, 2020-07-21:
Nice summary! I confirm there are less swallowed exceptions. These are really annoying

swamper123, 2020-07-22:
As well it is not such a hard style break anymore in the whole asyncio thematic. There have been so many differences and problems in <=Python3.6 Versions about compatibility and using, which seems to be solid since 3.6 upwards.
So in my opinion we would profit if we add these tests as well.

HarrySky, 2020-07-23:
@swamper123 `asyncio.Lock` is not deprecated, checked and only `loop` parameter is deprecated since 3.8 :confused: 

Extending Travis is a great idea :+1:

swamper123, 2020-07-27:
@HarrySky I just had a look and you were right. Fixed post. :)

closed 2020-07-27 by swamper123