FreeOpcUa/opcua-asyncio #246: 0.8.3 -> 0.8.4 changes?

AndreasHeine, 2020-07-21:
0.8.4:
[image]
0.8.3
[image]

!?

oroulet, 2020-07-21:
Must be something in xml... Needs debugging ;-)

AndreasHeine, 2020-07-21:
those changes.... -_-°

oroulet, 2020-07-22:
prosys browser cannot browse DataTypes on opcua-asyncio... might be the same issue...

oroulet, 2020-07-22:
The bad commit is https://github.com/FreeOpcUa/opcua-asyncio/commit/b0b54cb954c7bd037ff43e1e5236e87a256e8db4  @swamper123 @AndreasHeine

oroulet, 2020-07-22:
arg wrong. I cheked manually all commits and it seems that first break is the next commit https://github.com/FreeOpcUa/opcua-asyncio/commit/47920a7fe022a8682c833ff99124273943bf3982

oroulet, 2020-07-22:
bug here: https://github.com/FreeOpcUa/opcua-asyncio/commit/47920a7fe022a8682c833ff99124273943bf3982#diff-373971016c019c1ed89fd42c9829acb8R394

AndreasHeine, 2020-07-22:
> bug here: 47920a7#diff-373971016c019c1ed89fd42c9829acb8R394

 might explain -> #247

AndreasHeine, 2020-09-17:
@oroulet looks like the error is back!?

closed 2020-09-17 by AndreasHeine