FreeOpcUa/opcua-asyncio #247: error instantiate method -> dname -> str

AndreasHeine, 2020-07-21:
my time is a little limited this week maybe someone else could have a look?

```python
await instantiate(objects, server.get_node(f"ns=0;i=58"), ua.NodeId.from_string(f"ns={idx};i=1500"), f"{idx}:Obj1", "Obj1", idx, instantiate_optional=True) # dname="Obj1" => AttributeError: 'str' object has no attribute 'ua_types'
```

Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\server\uaprocessor.py", line 119, in process_message 
    return await self._process_message(typeid, requesthdr, seqhdr, body)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\server\uaprocessor.py", line 209, in _process_message
    self.send_response(requesthdr.RequestHandle, seqhdr, response)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\server\uaprocessor.py", line 46, in send_response    
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 241, in struct_to_binary      
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in list_to_binary        
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 269, in to_binary
    return struct_to_binary(val)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 241, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 269, in to_binary
    return struct_to_binary(val)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 246, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 259, in to_binary
    return pack_uatype(vtype, val)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 179, in pack_uatype
    return struct_to_binary(value)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 238, in struct_to_binary
    for name, uatype in obj.ua_types:
AttributeError: 'str' object has no attribute 'ua_types'
Error while processing message
Traceback (most recent call last):
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\server\uaprocessor.py", line 119, in process_message
    return await self._process_message(typeid, requesthdr, seqhdr, body)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\server\uaprocessor.py", line 209, in _process_message
    self.send_response(requesthdr.RequestHandle, seqhdr, response)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\server\uaprocessor.py", line 46, in send_response
    struct_to_binary(response), message_type=msgtype, request_id=seqhdr.RequestId)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 241, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 269, in to_binary
    return struct_to_binary(val)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 241, in struct_to_binary
    packet.append(list_to_binary(uatype[6:], val))
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in list_to_binary
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 281, in <listcomp>
    pack = [to_binary(uatype, el) for el in val]
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 269, in to_binary
    return struct_to_binary(val)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 246, in struct_to_binary
    packet.append(to_binary(uatype, val))
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 259, in to_binary
    return pack_uatype(vtype, val)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 179, in pack_uatype
    return struct_to_binary(value)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python37\lib\site-packages\asyncua\ua\ua_binary.py", line 238, in struct_to_binary
    for name, uatype in obj.ua_types:
AttributeError: 'str' object has no attribute 'ua_types'

AndreasHeine, 2020-07-22:
forgot to say it chrashes on browsing...

swamper123, 2020-12-16:
@AndreasHeine Is this still a thing?

AndreasHeine, 2020-12-16:
yes still in 0.9.3

```python
import asyncio
from asyncua import Server, ua
from asyncua.common.instantiate_util import instantiate

async def main():
    server = Server()
    await server.init()
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    await instantiate(
        server.get_objects_node(), 
        server.get_node(f"ns=0;i=58"), 
        ua.NodeId.from_string(f"ns={idx};i=1500"), 
        f"{idx}:Obj1", 
        "Obj1", 
        idx, 
        instantiate_optional=True
        ) # dname="Obj1" => AttributeError: 'str' object has no attribute 'ua_types'

    async with server:
        while 1:
            await asyncio.sleep(1)

asyncio.run(main())
```