FreeOpcUa/opcua-asyncio #250: opcua-asyncio subscription and XML generation

SeongsuByeon, 2020-07-31:
Dear all, 

I just started to look at opcua-asyncio for the project and need advice on it. 
My goal is to read values from defined(monitored) nodes and write these values in xml format. 
I am using subscription to get updated values from the MDE server. 

Questions
1. I have to output XML data with very high frequency (10Hz) for our another application. But callback function is called when the client received the data change message.  In this case, subscription is not a efficient way? is it better to get_values with node information without subscription? 

 2. I have to generate XML while subscribing the monitored nodes in server. I tried to put while loop inside async main function. But this while loop blocks others. What would be the efficient way to generate XML data with updated values from call back function? 

3. I think it is very important to use callback function efficiently to get values from nodes with one subscription. 
Is there any example code to look how "datachange_notification" function is used for getting and saving values? 

Its quite challenging for me to come up with clever idea. 
please give me an advice on this.

swamper123, 2020-08-03:
> 1. I have to output XML data with a very high frequency (10Hz) for our other application

This is a problem, we may not handle in this scope. Gaining Data via Client and export Nodes (not Values!) is what we do (please correct me if I'm wrong). There isn't an method for "valueexport" (not that I know). So here you are on your own how you implement the Client in your project. As well I don't know how fast a value can be wrapped in XML, this would depend on the amount of data and the efficiency of your function. Perhaps the lxml library may provide some stuff to help you out.

> 2. I have to generate XML while subscribing to the monitored nodes in server. I tried to put a while loop inside the async main function. But this while loop blocks others

If you had an `await asyncio.sleep(mytimevar)` it shouldn't block anything in the eventloop, except you deal with multiple eventloops.

SeongsuByeon, 2020-08-12:
@swamper123  Thank you for your explanation. 

`            nodes_1 = [CourseOverGround_OS,Heading_OS,Latitude_OS,Longitude_OS,RateOfTurn_OS]
                     
            nodes_2 = [SpeedOverGround_OS,SpeedThroughWater_OS,FwdEot,FwdRudder,FwdDriving,AftEot,AftRudder,AftDriving,
                     communicationType]
        
            await subscription.subscribe_data_change(nodes_1)
            await asyncio.sleep(1000)
            await subscription.subscribe_data_change(nodes_2)
            await asyncio.sleep(1000)`

swamper123, 2020-08-18:
May I ask why you have got asyncio.sleep there anyway?
You can drop a list of Nodes to `subscription.subscribe_data_change` if you want subscribe more than one.

SeongsuByeon, 2020-08-18:
@swamper123  I forgot to mention that I have several objects(each objects has several nodes) that I have to subscribe.
As you can see below, there are many objects (ex: M34593) that represent each ship's identification number. And each ship contains many variables (dynamic and static). So the challenging part is not to subscribe several nodes(variables) of a specific ship (M235223) but to subscribe all these ships at the same time. Thats why I used multi-threading in pyhton opcua.  I know that it is not clean way but it is working. Is it possible to implement this without multi-threading in opcua-asyncio?

[image]

swamper123, 2020-09-14:
Sorry for the late response, it was going under in my work. 😞 
I haven't done such a thing yet, but perhaps you can use a method to check the amount of ships and use a [process Pool](https://docs.python.org/3/library/asyncio-eventloop.html#executing-code-in-thread-or-process-pools) to subscribe for each ship. Perhaps you can pass Client as a parameter to all tasks in the pool and subscribe to them.
How "fast" is your realtime frame? ms? s? min?
Or have you already figured out another way?

closed 2023-03-29 by schroeder-