FreeOpcUa/opcua-asyncio #252: Communication between Prosys Server and opcua-asyncio client generates error

engmsilva, 2020-08-05:
Scenario

I'm using the Prosys simulation server and a pc and running the opcua-asyncio client on another pc. Through opcua clients and Prosys for android I can communicate with the Prosys server, so there is no communication problem on the network. When I try to make the same communication with opcua-asyncio it generates the error below. How can I resolve this error? I'm using this sample code:

``
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/client_to_prosys.py
``
As the code in the example is using the same default parameters as the server, I didn't change it, I just changed the url that points to the server. I found some suggestions for one of the errors (BadTcpMessageTooLarge) to increase the maximum message size, but it doesn't show how to do this.

Another question in the code is about the stretch (line 31 of the example file) below.

``
static_idx = await client.get_namespace_index ("http://www.prosysopc.com/OPCUA/StaticNodes")
``

 The url is the same as the one declared in the Prosys server namespaces, but when placed in the browser address bar it returns a 404 error. Is this right?

``
WARNING:asyncua.uaprotocol:Received an error: MessageAbort(error:StatusCode(BadTcpMessageTooLarge), reason:Bad_TcpMessageTooLarge (code=0x80800000, description="Chunk size (132) exceeded maximum (-1)"))
CRITICAL:asyncua.client.ua_client.UASocketProtocol:Received an error: MessageAbort(error:StatusCode(BadTcpMessageTooLarge), reason:Bad_TcpMessageTooLarge (code=0x80800000, description="Chunk size (132) exceeded maximum (-1)"))
ERROR:asyncua.client.ua_client.UASocketProtocol:Exception raised while parsing message from server
Traceback (most recent call last):
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 159, in _call_callback
    self._callbackmap[request_id].set_result(body)
KeyError: 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 80, in _process_received_data
    self._process_received_message(msg)
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 99, in _process_received_message
    self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 161, in _call_callback
    raise ua.UaError(
asyncua.ua.uaerrors._base.UaError: No request found for request id: 0, pending are dict_keys([1])
WARNING:asyncua.client.ua_client.UaClient:disconnect_socket was called but connection is closed
Traceback (most recent call last):
  File "/home/marcelo/Documentos/OPC UA/python/opcua_asyncio.py", line 72, in <module>
    asyncio.run(main())
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/marcelo/Documentos/OPC UA/python/opcua_asyncio.py", line 27, in main
    async with Client(url=url) as client:
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/client.py", line 71, in __aenter__
    await self.connect()
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/client.py", line 228, in connect
    await self.open_secure_channel()
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/client.py", line 278, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 274, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/home/marcelo/.virtualenvs/OPC_UA-HBbcF4fs/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 203, in open_secure_channel
    result = await asyncio.wait_for(
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError
``

AndreasHeine, 2020-08-05:
> The url is the same as the one declared in the Prosys server namespaces, but when placed in the browser address bar it returns a 404 error. Is this right?

opc ua is domaindriven so its just a name not a website!

prosys always makrs trouble... for the messagesize thing try to run client and server on different machines!

engmsilva, 2020-08-05:
> O prosys sempre cria problemas ... para a coisa do tamanho da mensagem, tente executar o cliente e o servidor em máquinas diferentes (verifique as configurações do firewall)!

oroulet, 2020-08-05:
I have a fix for that one in my branch. Need to merge...

oroulet, 2020-08-05:
Why did you close the issue?

engmsilva, 2020-08-05:
> Por que você encerrou o problema?

> Por que você encerrou o problema?

I must have closed it by mistake. You can open it again.

engmsilva, 2020-08-05:
> Eu tenho uma correção para esse na minha filial. Precisa mesclar ...

It would be nice if I could make it available, as I am trying to establish communication between server and client for hours.

oroulet, 2020-08-05:
in fact it is on master. Just clone master and install

engmsilva, 2020-08-08:
After a while reading a node in a loop in the background through a thrend, it is generating a timeout error.

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "main.py", line 34, in run
    asyncio.run(self.opc_ua())
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "main.py", line 28, in opc_ua
    value = await var.read_value()
  File "/home/marcelo/Documentos/OPC UA/flask-socketio/opcua-asyncio/asyncua/common/node.py", line 183, in read_value
    result = await self.read_data_value()
  File "/home/marcelo/Documentos/OPC UA/flask-socketio/opcua-asyncio/asyncua/common/node.py", line 194, in read_data_value
    return await self.read_attribute(ua.AttributeIds.Value)
  File "/home/marcelo/Documentos/OPC UA/flask-socketio/opcua-asyncio/asyncua/common/node.py", line 301, in read_attribute
    result = await self.server.read(params)
  File "/home/marcelo/Documentos/OPC UA/flask-socketio/opcua-asyncio/asyncua/client/ua_client.py", line 360, in read
    data = await self.protocol.send_request(request)
  File "/home/marcelo/Documentos/OPC UA/flask-socketio/opcua-asyncio/asyncua/client/ua_client.py", line 141, in send_request
    data = await asyncio.wait_for(
  File "/home/marcelo/.pyenv/versions/3.8.5/lib/python3.8/asyncio/tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

bobocha214, 2023-12-14:
Have you solved it？ I also encountered this kind of problem, I tried to use the reconnection operation, the connection to the local address will have a reconnection message, but the connection to the external opcua address will have this error.