FreeOpcUa/opcua-asyncio #253: Setting security policy without loading private key and certificate

qorinator, 2020-08-05:
Hi,

I am new in using opcua, so bare with the question.

Currently I'm implementing a python script for connecting with a opcua device. The device is configured to allow connection with Basic256 signed and encrypted.

In this library I saw, to set the security I need to load both private key and certificate file. 

I talked to the engineers that work with the opcua device and they told me that the server in the device has a certificate and a private key, the client(python script) needs to accept this certificate and use 'anonymous' as user. This is how they work as well with uaExpert. 

If I don't set any security policy, I will receive this
```
CRITICAL:asyncua.client.ua_client.UASocketProtocol:Received an error: MessageAbort(error:StatusCode(BadSecurityPolicyRejected), reason:None)
```

Questions: 
* Is there a way to set the security policy without loading the certificate and private key files? and to use the certificate from the server?
* Is this a 'normal' way of working with opcua ?

Thanks!

AndreasHeine, 2020-08-05:
could you post your client security string?

AndreasHeine, 2020-08-05:
to be more precise if you want to use Sign&Encrypt you will need a Cert and a Key otherwise its not possible to Sign or Encrypt!

with open ssl you can generate your own x509v3:

> 
> Generate your own Certificate
> Step 1: Change ssl.conf (subjectAltname, country, organizationName, ...)
> Step 2: openssl genrsa -out key.pem 2048
> Step 3: openssl req -x509 -days 365 -new -out certificate.pem -key key.pem -config ssl.conf

content of ssl.conf:
```
[ req ]
default_bits = 2048
default_md = sha256
distinguished_name = subject
req_extensions = req_ext
x509_extensions = req_ext
string_mask = utf8only
prompt = no

[ req_ext ]
basicConstraints = CA:FALSE
nsCertType = client, server
keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage= serverAuth, clientAuth
nsComment = "OpenSSL Generated Certificat"
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
subjectAltName = URI:urn:opcua:python:server,IP: 127.0.0.1

[ subject ]
countryName = DE
stateOrProvinceName = HE
localityName = HE
organizationName = AndreasHeine
commonName = PythonOpcUaServer
```

closed 2020-08-05 by oroulet

AndreasHeine, 2020-08-06:
ok! i need a screenshot (UaExpert) on which i can see which endpoints are avalible on your server!

qorinator, 2020-08-06:
This? 

[image]

AndreasHeine, 2020-08-06:
could you try this code:

its the normal python-opcua.... but you can use the sync wrapper for async ua or just test it with the none asyncio module! or just add the missing await statements

```python
from opcua import Client, ua

url = "opc.tcp://127.0.0.1:4840"
client = Client(url)

client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")
client.application_uri = "urn:example.org:FreeOpcUa:python-opcua"

client.connect()
print(f"Connected to: {url}")

servicelevel_node = client.get_node("ns=0;i=2267")

servicelevel_node_value = servicelevel_node.get_value()
print(servicelevel_node_value)

client.disconnect()
```

you can use your cert and key or those from this repo https://github.com/AndreasHeine/SecurePythonOpcUaClient/tree/master/x509v3

qorinator, 2020-08-06:
it works!
I am guessing your certificate was generated based on the ssl.conf that is checked in. 

how did you generate the certificate and key that were checked in your repo? I was following the instruction mentioned in this file https://github.com/FreeOpcUa/python-opcua/blob/master/examples/generate_certificate.sh

AndreasHeine, 2020-08-06:
https://github.com/AndreasHeine/SecurePythonOpcUaServer/blob/master/README.md

qorinator, 2020-08-07:
thank you so much for your help! 

just for my understanding what is the purpose of having the certificate and private key for the client? and how will the server recognize these ?

AndreasHeine, 2020-08-07:
it similar to the ssl/tls handshake
https://upload.wikimedia.org/wikipedia/commons/a/ae/SSL_handshake_with_two_way_authentication_with_certificates.svg

but in opc ua are two main differences, a cert can be used to sign messages (message security) or the other thing would be to auth the client and get access to the server. So cert context can be auth or message security! There are many ways a opcua server can handle certs (it depends on the server and what he accept) python opcua accepts all certs so no deeper check is made at the moment (selfsigned certs are like a self printet passport -> easy to generate and not realy secure without a "CA - Certificate Authority" but if the message is encryptet others arent able to read them easily)

Further information about OPC UA https://reference.opcfoundation.org/v104/

qorinator, 2020-08-07:
Thank you so much for your answers and help! I think with this I can continue on.