FreeOpcUa/opcua-asyncio #257: Error with subscription datachange

HermannSt, 2020-08-13:
Pardon my ignorance, I am new to both Python and OPC UA, but I believe I have found a bug in the subscription module in either the SubHandler sample class  or the _call_datachange method of Subscription.
```python
class SubHandler:
    def datachange_notification(self, node, val, data):
        pass
```
I will take line 128 from subscription.py (At the time of writing, not sure if there's a way to link it) as an example, but it seems that all of _call_datachange would be affected:
```python
 self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
```
Seems to only have 3 parameters, which produces the error below. If I delete the self parameter from SubHandler.datachange_notification or add self to the method it works fine.

```python
ERROR:asyncua.common.subscription:Exception calling data change handler
Traceback (most recent call last):
  File "C:\Users\Hermann\AppData\Local\Programs\Python\Python38-32\lib\site-packages\asyncua\common\subscription.py", line 128, in _call_datachange
    self._handler.datachange_notification(data.node, item.Value.Value.Value, event_data)
TypeError: datachange_notification() missing 1 required positional argument: 'data'
```

AndreasHeine, 2020-08-13:
which version did you use?

HermannSt, 2020-08-14:
asyncua 0.8.4.
Python 3.8

I just made a new Pycharm project and made it with their virtual environment to test it on a clean install and the error is still there.

Here is my SubHandler and subscription code:
```python
import asyncio
import logging
from asyncua import Client, Node, ua
#import asyncua.common.ua_utils as ua_utils

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger('asyncua')

class SubHandler:

    def datachange_notification(self, node: Node, val, data):
        print("datachange_notification: NODE:%r VALUE:%s" % (node, val))
        #_logger.info('datachange_notification %r %s',  val, node)

    def event_notification(self, event):
        print("New event %r", event)

async def main():
    url = "opc.tcp://localhost:4840"
    async with Client(url=url) as client:

        uri = "HTL-DA"
        idx = await client.get_namespace_index(uri)

        # Subscription that requests data changes from node DA\Auto
        handler = SubHandler
        auto_node = await client.nodes.objects.get_child([f"{idx}:DA", f"{idx}:Auto", f"{idx}:MyVariable"])
        subscription = await client.create_subscription(500,handler)
        nodes = [
            auto_node,
            client.get_node(ua.ObjectIds.Server_ServerStatus_CurrentTime),
        ]
        sub_return = await subscription.subscribe_data_change(nodes)

        while True:
            await asyncio.sleep(10)

if __name__ == "__main__":
    asyncio.run(main())

```

Here is my full code just in case: 
```python
`import asyncio
import logging
from asyncua import Client, Node, ua
#import asyncua.common.ua_utils as ua_utils

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger('asyncua')

class SubHandler:

    def datachange_notification(self, node: Node, val, data):
        print("datachange_notification: NODE:%r VALUE:%s" % (node, val))
        #_logger.info('datachange_notification %r %s',  val, node)

    def event_notification(self, event):
        print("New event %r", event)

async def main():
    url = "opc.tcp://localhost:4840"
    async with Client(url=url) as client:
        _logger.info("Root node is: %r", client.nodes.root)
        _logger.info("Objects node is: %r", client.nodes.objects)

        # Node objects have methods to read and write node attributes as well as browse or populate address space
        _logger.info("Children of root are: %r", await client.nodes.root.get_children())

        uri = "HTL-DA"
        idx = await client.get_namespace_index(uri)
        _logger.info("index of our namespace is %s", idx)

        dataRoot = await client.nodes.objects.get_child(f"{idx}:DA")
        dataChildren = await dataRoot.get_children()

        #Print children name and value of DA
        for node in dataChildren:
            node_name = await node.read_display_name()
            print(f'NAME:{node_name.Text} ID:{node}')

            child = await node.get_children()
            for sensor in child:
                sensor_name = await sensor.read_display_name()
                print(f'\tNAME:{sensor_name.Text}\n\t\tID:{sensor}\n\t\tVALUE:{await sensor.read_value()}')

        # Subscription that requests data changes from node DA\Auto
        handler = SubHandler
        auto_node = await client.nodes.objects.get_child([f"{idx}:DA", f"{idx}:Auto", f"{idx}:MyVariable"])
        subscription = await client.create_subscription(500,handler)
        nodes = [
            auto_node,
            client.get_node(ua.ObjectIds.Server_ServerStatus_CurrentTime),
        ]
        sub_return = await subscription.subscribe_data_change(nodes)

        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())

```

AndreasHeine, 2020-08-14:
@HermannSt you are missing "()" after subjandler ;)

`handler = SubHandler()`

HermannSt, 2020-08-14:
haha...^^ well, thanks a lot, I really appreciated your help

closed 2020-08-14 by HermannSt