FreeOpcUa/opcua-asyncio #258: a little discovery

myahuang, 2020-08-15:
Hi, your web address is http://freeopcua.github.io which if I type it into network explorer, the web can't open ,when I  change the address to https://freeopcua.github.io, the web can properly opened.

AndreasHeine, 2020-08-19:
looks like a missing redirect (http -> https) on the github pages...

swamper123, 2020-08-19:
Edge and Chromium work as well. Perhaps a setting problem of your browser?

myahuang, 2020-08-19:
But it cant't opened by my Edge,Chromium and IE14. And I have closed firewall. OS is win10 enterprise

AndreasHeine, 2020-08-19:
china? maybe blocked! last time in china i used express vpn to get access to european sites.

swamper123, 2020-08-19:
I got win10 as well and the latest edge version and it works. So I guess this is still an IT Problem on your side somehow.

myahuang, 2020-08-19:
@AndreasHeine maybe. lol
@swamper123, if you all can access the web,it not be a problem. For me, may be for every members  in china,  the matter is just can't directly click the link to visit the web. We need manualy typed 'https' instead of 'http'.

closed 2020-08-19 by myahuang