FreeOpcUa/opcua-asyncio #259: why i can't connect ''server-with-encryption.py" with UaExpert,but i can connet the server with "client-with-encryption.py"

myahuang, 2020-08-15:
Hi!
   I can connect the server with client "client-with-encryption.py", then I add a server in UaExpert, and use the configure  same to client "client-with-encryption.py", but the application popup a warning dialog ,the warning message is " There was no exact match for the specified URL in the server's endpoints, using the first server certificate found for connecting".

AndreasHeine, 2020-08-15:
could you make a screenshot from ua expert and pls post the servers endpoint url plus configuration!

myahuang, 2020-08-15:
UaExpert configuration screenshot:
[image]
Because the default port 4840 have been occupied by other service ,so I change the server   ''server-with-encryption.py"  port to 14840 .Then configure the server Endpoint Url as 'opc.tcp://127.0.0.1:14840/freeopcua/server/'
Python console information is showing as following:
_"INFO:asyncua.server.binary_server_asyncio:New connection from ('127.0.0.1', 64367)
INFO:asyncua.server.uaprocessor:get endpoints request (None)
INFO:asyncua.server.internal_server:get endpoint
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('127.0.0.1', 64367)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('127.0.0.1', 64367), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('127.0.0.1', 64367)
INFO:asyncua.server.binary_server_asyncio:New connection from ('127.0.0.1', 64368)
INFO:asyncua.server.uaprocessor:Create session request (None)
INFO:asyncua.server.internal_session:Created internal session ('127.0.0.1', 64368)
INFO:asyncua.server.internal_session:Create session request
INFO:asyncua.server.internal_server:get endpoint
INFO:asyncua.server.uaprocessor:Activate session request (None)
INFO:asyncua.server.internal_session:activate session
ERROR:asyncua.server.uaprocessor:sending service fault response: "User does not have permission to perform the requested operation." (BadUserAccessDenied)
WARNING:asyncua.server.uaprocessor:User(role=<UserRole.Anonymous: 1>, name=None) attempted to do something they are not permitted to do
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('127.0.0.1', 64368)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('127.0.0.1', 64368), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('127.0.0.1', 64368)
INFO:asyncua.server.internal_session:close session ('127.0.0.1', 64368)
INFO:asyncua.server.subscription_service:delete subscriptions: []_

AndreasHeine, 2020-08-15:
is your server configured to use cert for auth?

myahuang, 2020-08-15:
My server is the example ''server-with-encryption.py",I only change this line 'server.set_endpoint("opc.tcp://0.0.0.0:_14840_/freeopcua/server/")'

AndreasHeine, 2020-08-15:
try Anonymus in ua expert auth settings for sign& encrypt ua expert generate a cert while installation!

myahuang, 2020-08-16:
@AndreasHeine Thanks for your replys! 
I have generated 5 files   in accordance with exampl  "generate_certificate.sh" successfully,I'm sorry for my pool knowledg of security policy ,could you tell what shall I do next?

AndreasHeine, 2020-08-16:
i dont understand your problem... you should be able to connect with ua expert if you choose the right authentication checkbox!

myahuang, 2020-08-18:
I'll do my best to grasp the librarys code , Thank you all the same!

closed 2020-08-18 by myahuang

myahuang, 2020-08-19:
I have solved the problem!
Step one  :use the UaExpert client to generate its certificate as name uaexpert.der.
Step two  :Copy the certificate to the ''server-with-encryption.py" folder "certificates".
Step three: Add a line code :"await cert_user_manager.add_admin("certificates/uaexpert.der", name='tiger')" to replace the former one.
Step four: In the UaExpert client server setting, Set the same securitu setting as ''server-with-encryption.py",In the auth setting options ,select 'Anonymous'
The key point is that you must use UaExpert to generate its certificate ,then you  add code to  trust the certificate  in the server program