FreeOpcUa/opcua-asyncio #260: OPCUA Security

Jayant-Pancholi, 2020-08-17:
I am trying to Write OPCUA Server featuring Secuirty .

Few queries -
Where does the Server stores its Own certificate ?
Where does the Server stores Trusted and Rejected Certificate ?
I am Choosing one particular type of certificate and encryption , can I still have access to that server using None as well ?
Can I have user specific Nodes and Variables ? For eg - Particular user will have access to particular Nodes of Server in address space ?

AndreasHeine, 2020-08-17:
> Where does the Server stores its Own certificate ?

It depends on what path you give in "load_certificate" and "load_private_key"

> Where does the Server stores Trusted and Rejected Certificate ?

at the moment we accept all certs (Encryption) if you use certs for auth (hevent used that)

> I am Choosing one particular type of certificate and encryption , can I still have access to that server using None as well ?

```python
 def set_security_policy(self, security_policy, permission_ruleset=None):
        """
        Method setting up the security policies for connections
        to the server, where security_policy is a list of integers.
        During server initialization, all endpoints are enabled:
                security_policy = [
                            ua.SecurityPolicyType.NoSecurity,
                            ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
                            ua.SecurityPolicyType.Basic256Sha256_Sign
                                ]
        E.g. to limit the number of endpoints and disable no encryption:
                set_security_policy([
                            ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt])
        """
        self._security_policy = security_policy
        self._permission_ruleset = permission_ruleset

    def set_security_IDs(self, policy_ids):
        """
            Method setting up the security endpoints for identification
            of clients. During server object initialization, all possible
            endpoints are enabled:
            self._policyIDs = ["Anonymous", "Basic256Sha256", "Username"]
            E.g. to limit the number of IDs and disable anonymous clients:
                set_security_IDs(["Basic256Sha256"])
        (Implementation for ID check is currently not finalized...)
        """
        self._policyIDs = policy_ids
```
security policy and security ids are lists so all containing options will be avalible on this server!

> Can I have user specific Nodes and Variables ? For eg - Particular user will have access to particular Nodes of Server in address space ?

no thats not implemented right now... i read the spec. part but most vendors have not implemented it atm or are still working on it. the only solution for now would be a different server instence on a different port!

Jayant-Pancholi, 2020-08-17:
I have written my own server as follows - 

```
async def main():

    cert_user_manager = CertificateUserManager()
    await cert_user_manager.add_user("peer-certificate-example-3.der", name='test_user')

    server = Server()

    await server.init()

    server.set_endpoint("opc.tcp://192.168.43.36:4843/freeopcua/server/")
    server.set_security_policy([ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt])
    # load server certificate and private key. This enables endpoints
    # with signing and encryption.

    await server.load_certificate("peer-certificate-example-1.der")
    await server.load_private_key("peer-private-key-example-1.pem")
```
This is following 'Basic256Sha256' type of security.
Once I try connecting to any client. Client will also send a copy of its certificate.
Where will the Server going to store that certificate at the server end.
Also as per my understanding Server also has to accept it and put it under trusted certificates.
On what basis the Client certificate get into Trusted / Rejected on the Server End. and where can I see this.

AndreasHeine, 2020-08-17:
@JoeyFaulkner 🤔 guess thats a question for you! havent used that yet.

AndreasHeine, 2020-08-18:
> Where will the Server going to store that certificate at the server end.

https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/server/user_managers.py

```python
class CertificateUserManager:
    """
    Certificate user manager, takes a certificate handler with its associated users and provides those users.
    """
    def __init__(self):
        self._trusted_certificates = {}

    async def add_role(self, certificate_path: str, user_role: UserRole, name: str, format: str = None):
        certificate = await uacrypto.load_certificate(certificate_path, format)
        if name is None:
            raise KeyError

        user = User(role=user_role, name=name)

        if name in self._trusted_certificates:
            logging.warning(f"certificate with name {name} "
                            f"attempted to be added multiple times, only the last version will be kept.")
        self._trusted_certificates[name] = {'certificate': uacrypto.der_from_x509(certificate), 'user': user}

    def get_user(self, iserver, username=None, password=None, certificate=None):
        if certificate is None:
            return None
        correct_users = [prospective_certificate['user'] for prospective_certificate in self._trusted_certificates.values()
                         if certificate == prospective_certificate['certificate']]
        if len(correct_users) == 0:
            return None
        else:
            return correct_users[0]

    async def add_user(self, certificate_path: str, name: str, format: str = None):
        await self.add_role(certificate_path=certificate_path, user_role=UserRole.User, name=name, format=format)

    async def add_admin(self, certificate_path: str, name:str, format: str = None):
        await self.add_role(certificate_path=certificate_path, user_role=UserRole.Admin, name=name, format=format)
```

AndreasHeine, 2020-08-18:
did you `set_security_IDs(["Basic256Sha256"])`?

JoeyFaulkner, 2020-08-18:
hey! Yes as far as I can tell you're very close to this working, all you need is to change

`    server = Server()`
to 
`    server = Server(user_manager=cert_user_manager)`

such as here: 
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-with-encryption.py#L20

does this help at all?

Jayant-Pancholi, 2020-08-18:
@JoeyFaulkner I tried doing this but it says unexpected argument to the Server Class

JoeyFaulkner, 2020-08-18:
hmm.. its a relatively recent change, is it possible you're using an older version of asyncua? For example here you can see that user manager is an argument to the Server class :)

https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/server/server.py#L76

Jayant-Pancholi, 2020-08-18:
Yes I guess this is altogether a different instance of server class itself.
But I can see again that there are other many changes from the same.
Will it affect any other features and functionalities too ?

JoeyFaulkner, 2020-08-18:
its hard to say.. if you use virtualenv (https://virtualenv.pypa.io/en/latest/) you can try installing a newer version and seeing if your code still works after updating it :)

JoeyFaulkner, 2020-08-18:
> I have written my own server as follows -
>
> ```
> async def main():
>
>     cert_user_manager = CertificateUserManager()
>     await cert_user_manager.add_user("peer-certificate-example-3.der", [...]

the code you've linked in this comment will have the same functionality (i believe)

Jayant-Pancholi, 2020-08-19:
@JoeyFaulkner is the latest version of asyncua merged and made available with pip ?
Or do I manually have to use the updated server.py as mentioned above ?

JoeyFaulkner, 2020-08-19:
im not sure, i would guess since you're having this problem it isnt. You can install directly with pip from a github project using this method https://www.freecodecamp.org/news/how-to-use-github-as-a-pypi-server-1c3b0d07db2/

Jayant-Pancholi, 2020-08-20:
@JoeyFaulkner While setting up a user, where am I setting up its password in the example code.

Jayant-Pancholi, 2020-08-21:
For better understanding of my current implementation, Please refer attached image.

Its to understand security part better in all three aspects -
Anonymous
User Auth.
certificate and Private Key combination

[
[image]
](url)

AndreasHeine, 2020-08-21:
if you auth with a certificate then you dont need a username and password the cert is you access key if you want just a basic password protectet server:

```python
user_db = {
    "user1": "pw1"
}

...

def user_manager(isession, username, password):
    isession.user = UserManager.User
    return username in users_db and password == users_db[username]

...

server.user_manager.set_user_manager(user_manager)
```

JoeyFaulkner, 2020-08-21:
@AndreasHeine 's code will work for an example project and if you just want to get something up and running then this makes sense!

But just to let you know, if you want to have username/pw functionality in production then storing the password in plaintext is bad form, and long term in this library we should implement salting/hashing functionality for passwords. I think its on the roadmap :).

ffoehlisch, 2020-09-11:
Hello,

i'm just trying to port some projects from opcua to asyncua. I'm using the actual git master HEAD from 2020-09-05 (sha=f3de377a8fa8be11481...) with python 3.8.0.

My aim is to enable username/password authentication and currently I'm struggeling with the changed authentication mechanism, at least when security is on (Basic256Sha256 SignAndEncrypt).

I Inherited a UserManager and implemented a get_user() method for it. All works fine and the method is called upon an authentication request from my client (UaExpert) but it seems like the password comes encrypted: 
This: 
b'\x1d\x9e\x9d8\xf9V\x13\xd0H%\x18\x08\xfb\x87_\xf1k"\xa7\x03%"\xbf"9R\x9f0\xbd\xdcn\x88\xdb\n\xfc\x04\x9e\x13\x04\xdc\xa2\xd2>t_\x08fJ\x95\x12\x9b\x02)\x8aPp\xaatC\xb9\x86\xd4\x8d\x15~\xba\x87\x88\x9b\\g\x91N\x03\xac\xb3\xae\xbc\xe6S\xf7\x08\xcaVkI\n /\xdf\x144m\xfa7\x04\xc5\x83\x9d\xe63\x81#4\x190>r\xa51\x94\x91\x95\xd0\xde\x93\xc3e\xdaz\x00\xcf\x0fW\x85\x86\x91.\x8b\x85\xaax\x9e*\x8d\x12<N=\xa2,\xcc\xd3\x01\xe7\x027\xb8h\xec@\xcb\ny\xd4\r*\x9bam\xd1\x8bBT\xdbi9\x95+\xfb\xc5H\x08\xb7\xfd\xea\xa6b\xcf\xbc$\x91\xa3Il\xbb\xd8\xb2\xaf\x06\x13\xde-\xdd\xf4\x96\xc4\xd6G\x99\x87c=K\xb2\x9fq\xcb\\!\xee\xce\x8e\x1a%\xaa\x1e=J\x13\xc9\xe8\xb4g\x8e\xc4\xacW\xb6\xceU\xc1x>\x01\x06\xa1\xa8h:V\xe6\xcbI\x0c\xe0}\xfd\xb0\xdd\x95\x95\xf8\x04(\x90'

must somehow contain 'dummy_password'. 

Can you help with what needs to be done to decrypt the password in order to verify it?

Please note that after I found this issue thread I tried to do as Andreas Heine suggested: 
` server.user_manager.set_user_manager(user_manager) `

However,  it seems like the UserManager doesn't have a user_manager attribute (at least in the HEAD version).

Here my code: [server_asyncua.py.txt](https://github.com/FreeOpcUa/opcua-asyncio/files/5208803/server_asyncua.py.txt)
And here the console output from the server for an authentication attempt: [server_asyncua.py.log](https://github.com/FreeOpcUa/opcua-asyncio/files/5208831/server_asyncua.py.log)

closed 2023-03-29 by schroeder-