FreeOpcUa/opcua-asyncio #262: AttributeError: 'Node' object has no attribute 'get_browse_name'

DejonckheereWard, 2020-08-18:
I have been getting this error with version 8.3 and higher, without deprecation warning or anything. Searched for a while and found that the name was changed to "read_browse_name()" instead of "get_browse_name".
Hasn't updated in the [ReadTheDocs ](https://python-opcua.readthedocs.io/en/latest/_modules/opcua/common/node.html#Node.get_browse_name)page either, any reason why this was changed?

https://github.com/FreeOpcUa/opcua-asyncio/blob/f653d611b210f5a4f8b285382ad6a345f5e30ed9/asyncua/common/node.py#L82

AndreasHeine, 2020-08-18:
hould be closer to spec. names i guess.

DejonckheereWard, 2020-08-18:
I see, this does mean it has a different name than the original python-opcua. 
Not relevant for anymore but possible to update docs or add a deprecation warning?

AndreasHeine, 2020-08-18:
i thought there was a method reference to the old name... maybe its got removed!

`get_browse_name = read_browse_name`

DejonckheereWard, 2020-08-18:
Yeah, couldn't find a reference either, hope it gets updated.

closed 2021-08-17 by DejonckheereWard