FreeOpcUa/opcua-asyncio #265: How to generate events with client

juliuskittler, 2020-08-19:
Dear `opcua-asyncio` team, 

I have a question regarding events (as a newbie).

The server can create an event generator with the `get_event_generator` function and then use this generator to trigger events. However, the client does not have such a generator function. My question is: is there a way to trigger events with the client as well, which the server then listens to?

Some more background: I want the server to run a function whenever the client changes 3 variable values. For this, I intend to use a subscription handler on the client side. So far, I used the `datachange_notification` function in the subscription handler for this. However, the problem with this is that the the function on the server side is executed 3 times (when variable a, b and c change respectively). This is not what I want. Instead, I want to only run my function 1 time (when all three variables, a, b, and c have changed). Hence, my idea was to trigger an event by the client once this has happened (and let the server listen to this event instead of to variable value changes). I'd be super thankful for any advice on this.

Many, many thanks in advance for getting back.

AndreasHeine, 2020-08-19:
one way would be to subscribe serverside (server subscribes his own variables)

juliuskittler, 2020-08-19:
@AndreasHeine thanks for getting back- this might be helpful. 

May I ask what exactly you mean with changing the values "in one shot"? My current solution is to on the client side, change the 3 variables sequentially and let the server listen to adjustments in the variable that is changed **last** i.e. variable 3. For this, the server uses the `datachange_notification` handler function. (In the code below, simply assume that the three functions do indeed change the data of their corresponding variable each time they are called.)

```
await change_variable_1()
await change_variable_2()
await change_variable_3()
```

AndreasHeine, 2020-08-19:
client -> calls "opcua ua method" with the three values -> method sets the values serverside (all three values will be changed at once "in one shot") -> maybe triggers a server event if other clients have to be informes if they dont have subscribed these values and you should be done!

for example:
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-methods.py

swamper123, 2021-04-26:
Is your problem solved/still actual @juliuskittler ? Otherwise this case can be closed.

closed 2021-05-10 by oroulet