FreeOpcUa/opcua-asyncio #268: How do I identify the calling node in a method callback?

Randelung, 2020-08-20:
Hello!

As the title says, I'm trying to identify where a callback came from. I see from the source code that the arguments to the callback are the PARENT and the argument values. Project guidelines dictate several methods in one folder, so all methods would return the same parent ID.
Are you supposed to use a partial/lambda and supply a flag or are there other mechanisms to get the ID of the actual method node?

Or am I just misunderstanding OPC UA methods? :)

Also, is it possible to add/override the callback after node generation? I'd use `server.add_method_callback` again, is that safe?

Thanks for your help!

oroulet, 2020-08-20:
you cannot.
Also as far as I know using the caller to change the behviour of a method is bad progamming idea and is not possible in languages I know

Randelung, 2020-08-20:
I think there might be a misunderstanding.
Using the caller to change the behavior of a method? I'm trying to identify what method is being called in the first place. If I have two methods `root.add` and `root.multiply` and all I'm getting in my callback is `root` I don't know what to do...
Which is why I asked about my misunderstanding the OPC fundamentals - I find it hard to believe that it's impossible to find out which method was actually called on the OPC interface.
I'd have to implement a unique callback for each method... which is why I'm asking about partials/lambdas and what the intended usage is supposed to be.

oroulet, 2020-08-21:
Ok. I understand.  What is the issue of having two callbacks? Usually you want to map UA methods to some internal methods which are different anyway.
Also no problems in calling add_method_callbak many times but I have issues understanding why one would do that..

oroulet, 2020-08-21:
And yes the only think getting passed  to the method is the parent. No ideas why it is passed. I have never used it

AndreasHeine, 2020-08-21:
> No ideas why it is passed.

spec. wants it... its part of "Call Service Parameters" 
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.11.2/

objectId | NodeId | The NodeId shall be that of the Object or ObjectType on which the Method is invoked.In case of an ObjectType the ObjectType or a super type of the ObjectType shall be the source of a HasComponent Reference (or subtype of HasComponent Reference) to the Method specified in methodId.In case of an Object the Object or the ObjectType of the Object or a super type of that ObjectType shall be the source of a HasComponent Reference (or subtype of HasComponent Reference) to the Method specified in methodId.See OPC 10000-3 for a description of Objects and their Methods.
-- | -- | --
methodId | NodeId | NodeId   of the Method to invoke.If the objectId is the NodeId of an Object, it is allowed to use the NodeId of a Method that is the target of a HasComponent Reference from the ObjectType of the Object.

closed 2020-12-28 by oroulet