FreeOpcUa/opcua-asyncio #274: Identical browsenames in same Namespace?

swamper123, 2020-08-26:
Hey ho together,

I was wondering how a Client gets it's target Node in the end, if it doesn't know it's NodeID.
The only option is via BrowseName. But it is possible to create multiple Nodes with the same Browsename in the same Namespace, which can lead to chaos.

MyObject number one...
[image]
...and number two.
[image]

As I understand the [specs](https://reference.opcfoundation.org/v104/Core/docs/Part3/5.2.4/) here right, it is allowed to create multiple Nodes with the same Browsename, but they can't be in the same Node. 
They can be in the same namespace, but in different paths or they are in different namespaces.

So we may need to add a check before adding the Node to the server, if this Browsename already exists, in the Node where it should be added ( I hope this is understandable?).

Am I missinterpreting the specs?

oroulet, 2020-08-26:
yes it looks like we should accept several time the same bname as child of a ndoe

swamper123, 2020-12-07:
Solution provided in #306

swamper123, 2020-12-08:
Case solved and closed 🕵️

closed 2020-12-08 by swamper123