FreeOpcUa/opcua-asyncio #275: Event History and Example: "server-events-history.py"

AndreasHeine, 2020-08-28:
at first i have never used that history stuff ever before

1. 
```
Traceback (most recent call last):
  File "C:\Users\HeineAn\AppData\Local\Programs\Python\Python37-32\lib\site-packages\aiosqlite\core.py", line 153, in run
    result = function()
sqlite3.OperationalError: table 0_2253 has no column named MyNumericProperty
Historizing SQL Insert Error for events from NumericNodeId(i=2253): table 0_2253 has no column named MyNumericProperty
```

[image]

2. 
"server-events-history.py" is not async (looks like a copy of python-opcua)

I am reworking "2." facing issue "1.":
```python
import sys
sys.path.insert(0, "..")
import time
from datetime import datetime
import asyncio

from asyncua import ua, Server
from asyncua.server.history_sql import HistorySQLite

async def main():
    # setup our server
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    # get Objects node, this is where we should put our custom stuff
    objects = server.nodes.objects

    # populating our address space
    myobj = await objects.add_object(idx, "MyObject")

    # Creating a custom event: Approach 1
    # The custom event object automatically will have members from its parent (BaseEventType)
    etype = await server.create_custom_event_type(2, 'MyFirstEvent', ua.ObjectIds.BaseEventType,
                                            [('MyNumericProperty', ua.VariantType.Float),
                                             ('MyStringProperty', ua.VariantType.String)])
    # create second event
    etype2 = await server.create_custom_event_type(2, 'MySecondEvent', ua.ObjectIds.BaseEventType,
                                             [('MyOtherProperty', ua.VariantType.Float)])

    # get an event generator for the myobj node which generates custom events
    myevgen = await server.get_event_generator(etype, myobj)
    myevgen.event.Severity = 500
    myevgen.event.MyStringProperty = ua.Variant("hello world")
    myevgen.event.MyNumericProperty = ua.Variant(-456)

    # get another event generator for the myobj node which generates different custom events
    myevgen2 = await server.get_event_generator(etype2, myobj)
    myevgen2.event.Severity = 123
    myevgen2.event.MyOtherProperty = ua.Variant(1.337)

    # get an event generator for the server node which generates BaseEventType
    serverevgen = await server.get_event_generator()
    serverevgen.event.Severity = 111

    # Configure server to use sqlite as history database (default is a simple in memory dict)
    sqlite_history = HistorySQLite("my_event_history.db")
    await sqlite_history.init()
    server.iserver.history_manager.set_storage(sqlite_history)

    # starting!
    async with server:
        # enable history for myobj events; must be called after start since it uses subscription
        await server.iserver.enable_history_event(myobj, period=None)

        # enable history for server events; must be called after start since it uses subscription
        server_node = server.get_node(ua.ObjectIds.Server)
        await server.historize_node_event(server_node, period=None)
        count = 0
        while True:
            await asyncio.sleep(1)
            count += 0.1

            # generate events for subscribed clients and history
            await myevgen.trigger(message="This is MyFirstEvent " + str(count))
            await myevgen2.trigger(message="This is MySecondEvent " + str(count))
            await serverevgen.trigger(message="Server Event Message")

            await asyncio.sleep(1)

            # read event history from sql
            end_time = datetime.utcnow()
            server_event_history = await server_node.read_event_history(None, end_time, 0)
                

if __name__ == '__main__':
    asyncio.run(main())
```

AndreasHeine, 2020-08-28:
same issue in python-opcua example also does not run properly...
@zerox1212 @oroulet any suggestions?

AndreasHeine, 2020-08-28:
https://github.com/FreeOpcUa/python-opcua/blob/aed7ab0bbc8935a396b734634e378926a8dcbdf7/opcua/server/history_sql.py#L162

self._get_table_name(event.emitting_node) -> looks like "event.emitting_node" is always server !?

AndreasHeine, 2020-08-28:
```
print(myevgen)
print(myevgen2)
print(serverevgen)
```

> EventGenerator(Type:NumericNodeId(ns=2;i=2), Emitting Node:Node(NumericNodeId(ns=2;i=1)), Time:None, Message: LocalizedText(Encoding:0, Locale:None, Text:None))
> EventGenerator(Type:NumericNodeId(ns=2;i=5), Emitting Node:Node(NumericNodeId(ns=2;i=1)), Time:None, Message: LocalizedText(Encoding:0, Locale:None, Text:None))
> EventGenerator(Type:NumericNodeId(i=2041), Emitting Node:Node(NumericNodeId(i=2253)), Time:None, Message: LocalizedText(Encoding:0, Locale:None, Text:None))

zerox1212, 2020-08-28:
Emitting node is not always server, it's just the default. UA objects can also emit their own events, but it is not very common. Seems most library just emit all events from the server.\

The SQL implementation is rather simplistic. It makes a table for each node that can emmit and saves some of the event children in binary.

You need to look into how the table is being created when we ask to historize a node.

AndreasHeine, 2020-08-28:
`table 0_2253 has no column named MyNumericProperty` the thing i am wondering about is that it looks like the myobj events got emitted by server_obj and off course it does not have the necessary property (see db screenshot)...

zerox1212, 2020-08-28:
I'm trying to remember. We had problems with events a long time ago where only the server could emit them. I would have to search through old issues.

AndreasHeine, 2020-08-28:
mh, if it is like you said we maybe need to add something like event_source in addition to emitting_node but i will have a look at this weekend :)

zerox1212, 2020-08-28:
I will try to look back about what changed. Also if you are working on events you should be aware that it's built on internal subscriptions. This has some limitations and a real history implementation should probably not rely on subscription, but monitor the address space directly. I'm not sure.

zerox1212, 2020-08-28:
I found some unresolved stuff here: https://github.com/FreeOpcUa/python-opcua/issues/531

After reading some stuff from 2016 I am starting to remember that we were always getting all the custom properties in the subscription. So if you had multiple custom events on 1 object the subscript would return all the fields with a value of `None`.

Whatever has happened now is probably related. When the table get's made there are only columns for that event, but when the subscription triggers it will have ALL the fields and the SQL error arises.

zerox1212, 2020-08-28:
There was also a change in 2019 that could have broken this, but I didn't review the impact. https://github.com/FreeOpcUa/python-opcua/issues/896

araisch, 2020-08-28:
Rework in asnyc posted here:
#207 https://github.com/FreeOpcUa/opcua-asyncio/issues/207
But still broken..

zerox1212, 2020-08-28:
If you comment out the example to ONLY have a single event generator on the server node, does history work? Then try with ONLY events on a user object node and see if it works.

Events are very complex, I would try to get history to work with one case at a time instead of just making the example itself work.

AndreasHeine, 2020-08-28:
sure! but tomorow after breakfast in europe its evening ;)

AndreasHeine, 2020-08-29:
@zerox1212 found something now i can at least have one child event but second event in this node still a problem...!

```python
    def save_event(self, event):
        with self._lock:
            _c_sub = self._conn.cursor()
            # table = self._get_table_name(event.emitting_node) <---------- old
            table = self._get_table_name(event.get_event_props_as_fields_dict()["SourceNode"]._value) # <------new
            columns, placeholders, evtup = self._format_event(event)
            event_type = event.EventType  # useful for troubleshooting database

            # insert the event into the database
            try:
                _c_sub.execute(
                    'INSERT INTO "{tn}" ("_Id", "_Timestamp", "_EventTypeName", {co}) VALUES (NULL, "{ts}", "{et}", {pl})'
                    .format(tn=table, co=columns, ts=event.Time, et=event_type, pl=placeholders), evtup)

            except sqlite3.Error as e:
                self.logger.error('Historizing SQL Insert Error for events from %s: %s', event.emitting_node, e)

            self._conn.commit()

            # get this node's period from the period dict and calculate the limit
            period = self._datachanges_period[event.emitting_node]

            if period:
                # after the insert, if a period was specified delete all records older than period
                date_limit = datetime.utcnow() - period

                try:
                    _c_sub.execute('DELETE FROM "{tn}" WHERE Time < ?'.format(tn=table),
                                   (date_limit.isoformat(' '),))
                except sqlite3.Error as e:
                    self.logger.error('Historizing SQL Delete Old Data Error for events from %s: %s',
                                      event.emitting_node, e)

                self._conn.commit()

```

AndreasHeine, 2020-08-29:
need additional time on that...

AndreasHeine, 2020-09-03:
i am close to a solution... i am still trying to reduce some of the code (unnessesary function calls wich happen in sequence twice and did the same)!

AndreasHeine, 2020-09-03:
#279

closed 2020-10-02 by AndreasHeine