FreeOpcUa/opcua-asyncio #276: Security Vulnerability: history_sql.py

swamper123, 2020-08-28:
Like @AndreasHeine mentioned in [python-opcua](https://github.com/FreeOpcUa/python-opcua/issues/1118) there are possible SQL Injections in history_sql.py.

The lines I found are:

https://github.com/FreeOpcUa/opcua-asyncio/blob/e6146e8666a7c1ad43ccdf74cd5fad9d5605a2ba/asyncua/server/history_sql.py#L58

https://github.com/FreeOpcUa/opcua-asyncio/blob/e6146e8666a7c1ad43ccdf74cd5fad9d5605a2ba/asyncua/server/history_sql.py#L67

https://github.com/FreeOpcUa/opcua-asyncio/blob/e6146e8666a7c1ad43ccdf74cd5fad9d5605a2ba/asyncua/server/history_sql.py#L98

https://github.com/FreeOpcUa/opcua-asyncio/blob/e6146e8666a7c1ad43ccdf74cd5fad9d5605a2ba/asyncua/server/history_sql.py#L142

https://github.com/FreeOpcUa/opcua-asyncio/blob/e6146e8666a7c1ad43ccdf74cd5fad9d5605a2ba/asyncua/server/history_sql.py#L155

https://github.com/FreeOpcUa/opcua-asyncio/blob/e6146e8666a7c1ad43ccdf74cd5fad9d5605a2ba/asyncua/server/history_sql.py#L170

Because I neither worked with SQL before, nor understanding the better way Andreas mentioned, I may not provide a PR for that at the moment (I have to check on that on Monday again, I got to many other things in my mind, but maybe I will put that on my agenda.).

JoeyFaulkner, 2020-08-28:
this imo is a good first issue if anyone's interested, there is an explainer here (https://docs.python.org/3.4/library/sqlite3.html) how to do this with sqlite.

AndreasHeine, 2020-08-28:
@JoeyFaulkner its not only the query-params, in asyncua most are parametrized but the table-name is always f-string insertion and even if its possible or not to inject a evil-node name with identifier-type string it should be avoided at all costs! if someone builds a server with dynamic eventgeneration it could be potentialy possible to break into the db. the eventdata could be sensitiv or process critical and should be protected. we need to find a consisten solution for both python-opcua and asyncio!

JoeyFaulkner, 2020-08-28:
Ah dang! Good point. Thanks for the explanation

swamper123, 2021-01-06:
Would it be possible to add a "[little boby tables](https://imgs.xkcd.com/comics/exploits_of_a_mom.png)" in an node with a string identifier?

closed 2023-03-29 by schroeder-