FreeOpcUa/opcua-asyncio #277: security vulnerability: pickle module

swamper123, 2020-08-31:
The pickle module in Python is a hot iron. It is a good serialiser, but the danger to execute unwanted code exists.

Pickle is used here:

https://github.com/FreeOpcUa/opcua-asyncio/blob/3636b610ae17a4c318730399150431afb8a81d5e/asyncua/server/address_space.py#L590

There are multiple options:
- leave as is
- remove them
- add a logger warning & comments to be carful with that

What do you think about that?

(And would it be a good idea to add an extra Label for security vulnerabilities?)

swamper123, 2020-09-11:
*bump* 
No comments on that?

AndreasHeine, 2020-09-11:
sorry read it but too busy atm 😬 (still want to start with history sql rework)

swamper123, 2020-09-14:
@AndreasHeine No preassure! There are other contributers here as well with a comment on that topic, I guess/hope?! 😄

cbergmiller, 2020-09-15:
Are the `load_address_space` and `dump_address_space` used by anyone? Also note that these are blocking methods (will block the event loop). If they are not used i would vote to remove them.
If still needed then `AddressSpace.dump` and `load` could be refactored to serialize the node data as JSON.

swamper123, 2020-09-15:
I guess this was a "quick&dirty" method to load/dump the whole addressspace for test purposes perhaps? The clean way would be via the xml files.
That these things are blocking is reasonable, because you may pickle an invalid addresspace in the end if it doesn't "freeze" for that time.
A nice JSON dump/load would be nice, indeed.

swamper123, 2020-09-18:
I just found that `generate_address_space.py` is generating/using a pickle file. We may have to rework that as well then into a JSON alternative.

As well there seems to be an idea that you can load/dump the namespace in a faster way than via xml(? see LazyLoadingDict in address_space), perhaps we could extend that with saved values as well?

schroeder-, 2023-03-29:
pickel is disabled so closing this

closed 2023-03-29 by schroeder-