FreeOpcUa/opcua-asyncio #280: Does opcua-asyncio support structures with optional fields?

jfischer1, 2020-09-04:
I'm trying to connect to a server (< OPC UA 1.04) with AutoID companion spec structures via UA binary protocol with opcua-async 0.8.4. These structures contain optional fields. Seem like opc:Bits datatype somehow causes an error. Are optional structure fields supported by asyncua? If yes, perhaps you can give me a hint what I'm doing wrong? Thank you!

Here come the details of the error message:

When calling load_type_definition(), method fails with the following message
```
  File "...\site-packages\asyncua\ua\ua_binary.py", line 484, in struct_from_binary
    obj = objtype()
File "<string>", line 27, in __init__
AttributeError: module 'asyncua.ua' has no attribute 'Bit'
```
It looks like generated code is trying to use an non-existent type:
```
class RfidScanResult(object):
    ua_types = [
        ('LocationSpecified', 'Bit'),
        ('Reserved1', 'Bit'),
        ('CodeType', 'CharArray'),
        # ...

    def __init__(self):
        self.LocationSpecified = ua.Bit() # <-- This line causes the error
        self.Reserved1 = ua.Bit()
```

AndreasHeine, 2020-09-04:
> class RfidScanResult(object):
>     ua_types = [
>         ('LocationSpecified', 'Bit'),
>         ('Reserved1', 'Bit'),
>         ('CodeType', 'CharArray'),
>         # ...
> 
>     def __init__(self):
>         self.LocationSpecified = ua.Bit() # <-- This line causes the error
>         self.Reserved1 = ua.Bit()

where does this came from?

oroulet, 2020-09-04:
what version are you using? I tried implementing 1.04 structure stuff, at least in client code a few months ago. it is in master, maybe not released yet. importing xml files should also work..
you need to use load_data_type_definitions() instead of load_type_definitions()

might be some bugs though.... I did not tested that specific spec and I am quite busy currently

AndreasHeine, 2020-09-04:
Server: 
```python
import asyncio, random
from asyncua import ua, Server

async def main():
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://127.0.0.1:4840/test")
    obj = server.get_objects_node()
    
    """
    setup test namespace
    """
    await server.import_xml("Opc.Ua.Di.NodeSet2.xml")
    await server.import_xml("Opc.Ua.AutoID.NodeSet2.xml")

    async with server:
        print("started!")
        while 1:
            await asyncio.sleep(1)
            
if __name__ == "__main__":
    asyncio.run(main())
```

Client:
```python
import asyncio
from asyncua import Client, Node, ua

async def main():
    client = Client(url='opc.tcp://127.0.0.1:4840/test')
    async with client:
        result = await client.load_type_definitions()
        print(str(result))
        while 1:
            await asyncio.sleep(0)

if __name__ == "__main__":
    asyncio.run(main())
```

0.8.4: no error...

oroulet, 2020-09-05:
@AndreasHeine use load_data_type_definition() to load structs using the 1.04 definitions (code is much simpler, but both should work)

AndreasHeine, 2020-09-05:
@oroulet ah thats new there was something you changed but i did not looked at it yet 😉

jfischer1, 2020-09-07:
@oroulet Thank you for your quick response. When reading them, I noticed that I was a bit unprecise when describing the problem. The error does *not* occur when calling load_type_definitions(). I happens when reading a structured datatype value.

Regarding the used version: I'm using opcua-asyncio 0.8.4 at client side and a non- opcua-asyncio server with autoid datatypes as defined in AutoID Nodeset version 1.00.

jfischer1, 2020-09-07:
@AndreasHeine : Code is generated in general at structures.py@302:
```
    for element in model:
        code = element.get_code()
        exec(code, env)
    return env
```
whereas the code for structures is generated in structures.py@87.

oroulet, 2020-09-07:
@jfischer1 yes. So use the new method: load_data_type_definutiin() and use asyncua client from master

oroulet, 2020-09-07:
I want to know it there are still issues before releasing

jfischer1, 2020-09-07:
@oroulet I tried using master branch (f3de377) and the following code: 
```
        async with Client(url=self.url) as client:
            variable = client.get_node(nodeId)
            val1 = await variable.read_value()
            await client.load_data_type_definitions()
            val2 = await variable.read_value()
```
Results:
- No exception occurred. However, there are several warnings, probably because server does not provide DataTypeDefinition attributes:
```
2020-09-07 11:55:35 WARNING  Registring data type ns=3;i=3001 QualifiedName(3:ScanResult)
2020-09-07 11:55:35 WARNING  ns=3;i=3001 has no DataTypeDefinition atttribute
```
- I wonder if there should be any difference when looking at val1 and val2 in debugging view. Both only show the ExtensionObject as binary blog. Shouldn't this blob be parsed for val2 into members after load_data_type_definitions()? Or is this due to the missing server-side DataTypeDefintion attributes?

oroulet, 2020-09-07:
* 2020-09-07 11:55:35 WARNING  ns=3;i=3001 has no DataTypeDefinition atttribute

that error + the fact that you do only fet ExtensionObject after calling load_data_type_definitions() means that your server in reality does not implement 1.04...

then you have to use the old method load_type_definition() and yes in the old method, optional field may not be supported, I think there was a bug there...

oroulet, 2020-09-07:
OK now I see that your said that your server does not implement 1.04. so yes thre might be a bug there... I do not have time to look at it

closed 2021-07-19 by swamper123