FreeOpcUa/opcua-asyncio #286: git fetch fails bevuase of branch name

cbergmiller, 2020-09-15:
I can't fetch this repository any more and i think it is due to the use of the `>` character in a recently introduced branch.
Anyone else having similar trouble?

```
git fetch
error: cannot lock ref 'refs/remotes/origin/Fix-history_sql.py-emitting-node-is-not-the-source-->-sql-error': Unable to create 'PycharmProjects/opcua-asyncio/.git/refs/remotes/origin/Fix-history_sql.py-emitting-node-is-not-the-source-->-sql-error.lock': Invalid argument
From https://github.com/FreeOpcUa/opcua-asyncio
 ! [new branch]      Fix-history_sql.py-emitting-node-is-not-the-source-->-sql-error -> origin/Fix-history_sql.py-emitting-node-is-not-the-source-->-sql-error  (unable to update local ref)
```

swamper123, 2020-09-15:
[image]
This realy seems to be an error if you are using console commands (tested on Windows and Linux, same result via cmd). Perhaps UI stuff may handle it in another way.

AndreasHeine, 2020-09-16:
@swamper123 done!

swamper123, 2020-09-16:
@AndreasHeine I think it is still there. Have you pushed it?
[image]

AndreasHeine, 2020-09-16:
... some strange things happend must delete the branch!

cbergmiller, 2020-09-16:
Works again for me, thanks!

AndreasHeine, 2020-09-16:
sorry about that!

swamper123, 2020-09-16:
No worries, can happen. :)

closed 2020-09-16 by swamper123