FreeOpcUa/opcua-asyncio #288: Reconnecting to server that went away

Randelung, 2020-09-17:
Hi again!

I'm writing a proxy server to translate GUID requests to string requests since WinCC OA doesn't support the former.
Now I'm trying to have the client reconnect to the server if it goes down.

I'm testing the connection loss by terminating the WinCC OA OPC UA Server and seeing what happens.
So far I get an exception (0.8.4 from pip):
```
Traceback (most recent call last):
  File "/home/randelung/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 386, in _renew_channel_loop
    await self.open_secure_channel(renew=True)
  File "/home/randelung/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 278, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/home/randelung/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 274, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/home/randelung/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 204, in open_secure_channel
    self._send_request(request, message_type=ua.MessageType.SecureOpen),
  File "/home/randelung/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 130, in _send_request
    self.transport.write(msg)
AttributeError: 'NoneType' object has no attribute 'write'
```

I found #241, but I believe that the `AttributeError` would just be replaced by a `ConnectionError`.

I see that automatic reconnection is on the to-do list. Is there a way to e. g. subscribe to a connection loss event and reinitialize the client myself? That would allow me to restore subscriptions and refuse method calls on the proxy server.
I'd like to not change your code if possible, but my first idea would be to make `UASocketProtocol.state` a property and trigger a callback when it's being set to `None`.

Thanks!

AndreasHeine, 2020-09-17:
you need to monitor the connection with a cyclic read ode a node and if it fails you need to reconnect! all other methods are not robust enought...

Randelung, 2020-09-17:
OK, so I'm just supposed to let the exception happen?

AndreasHeine, 2020-09-17:
thats the sad part... there are many things which can break the connection and in the end you will always monitor the connection like all big vendors (like Simens) do

small hint how you could solve it:
https://github.com/AndreasHeine/opcua-sub-to-websocket/blob/9f664cd2d6d7f3225ca7334f049c2ff5069ad755/client.py#L62

Randelung, 2020-09-17:
OK, cool, thanks. I think that takes care of the followup questions like task disposal etc.

closed 2020-09-17 by Randelung

KishanKishore, 2021-03-16:
> OK, cool, thanks. I think that takes care of the followup questions like task disposal etc.

> you need to monitor the connection with a cyclic read of a node and if it fails you need to reconnect! all other methods are not robust enought...

@AndreasHeine is there any other method that does not depend on a node but throws a ```ConnectionError``` in case the connection is broken. For using the ```get_value``` method we are dependent on a valid node id. It would be simpler if we could bypass that dependency.

AndreasHeine, 2021-03-16:
@KishanKishore use the serverstate (ns=0;i=2259) or the servicelevel (ns=0;i=2267) all opcua server have these nodes!