FreeOpcUa/opcua-asyncio #295: ModuleNotFoundError: No module named 'dataclasses' on Python 3.6

danieljfarrell, 2020-09-22:
Hi folks,

Dataclasses were introduced in Python 3.7 but asyncua states that it supports Python 3.6. It is currently not usable on Python < 3.7.

Workaround is to install the backport,

    > pip search dataclasses
    dataclasses (0.7)                           - A backport of the dataclasses module for Python 3.6

A fix would be to include conditional in install requires,

```python
setup(
    # ...
    install_requires=[
        # ...
        "dataclasses; python_version<'3.7'",
    ]
)
```

swamper123, 2020-09-22:
Since @oroulet once mentioned to drop support for 3.6 in #272, I would pick that up to affirm such a step as well.
The workaround may work, but I would prefere to be on a stable supported >=3.7 side, instead of using a workaround. Especially since users.py is using it and not another "minor feature".

oroulet, 2020-09-22:
OK just pushed a commit to "fix" that. Especially the missing asyncio.run and create_tasks is always giving (small) issues

danieljfarrell, 2020-09-22:
That's great! Thanks.

closed 2020-09-22 by danieljfarrell